bogde/HX711 #114: esp32 multiple definition of `yield'

JAICHANGPARK, 2018-10-16:
i had compile error 

`multiple definition of `yield'`

amotl, 2019-02-17:
Hi there,

we just pushed some amendments to this popular library where we rebundled some pieces which have been loosely floating around before [1]. Most probably, your specific problem has been addressed by these updates. For more details, please follow the issue referenced above.

Nevertheless, we will be happy to hear about successful reports from the community running this on real iron, actually on as much as possible things still supported by the Arduino framework. Maybe you can afford some minutes if your MCU is still around?

Even if you have moved on to a next generation chip, there are chances the HX711 breakout will still be around and this library could make it work already. We will appreciate your feedback, thanks in advance.

With kind regards,
Andreas.

[1] https://github.com/hiveeyes/HX711/tree/spring-cleaning

JAICHANGPARK, 2019-02-19:
Hi  @amotl 
I was created that temporary hx711 library by myself to used my research. 
ESP32 was not problem to operated hx711.

Thanks for your interesting my Issue 

Regards
Dreamwalker

closed 2019-02-19 by JAICHANGPARK