bogde/HX711 #116: getUnits() or some other function to return value only after a full read

sundud, 2018-11-25:
Hi Bogde,

I am trying to use your library but one issue or enhancement that i would request is - either getUnits() or some new function return the 'final' weight added. 

To give a short background, the project is to add or remove weight in increments (rather than just weighing it once for a object). And i would like to get the updated weight (not the current intermediate weight increments that i currently get) doing a getUnits() while trying different inputs of 5,10,20. A delay also is not helping.

It is of importance to me to have this working, so needless to say i am ready to arrange compensations.

Thanks,
Sunil

bogde, 2018-11-28:
i'm not sure i understand what you want but i think you can do that using the current version. in fact, your example looks like you're already getting what you want, at least to my understanding of your use case.

maybe you can post an example of what you would actually expect to get.

sundud, 2018-11-28:
Sorry for not being clear - i wanted a method that returns the current value once it has stabilized. A recursive method resolved this. Thanks.

closed 2018-11-28 by sundud