bogde/HX711 #119: WiFiSecureClient crashes during connection

karesztrk, 2019-01-21:


amotl, 2019-01-21:
Hi there,

we added this part with #59 the other day. I believe nobody runs this ancient Arduino Core releases anymore, so this could well be removed. @el3ctrician also made an improvement here by #113 to let this read like:
```
#ifndef ESP8266
    #if ARDUINO_VERSION <= 106
        // "yield" is not implemented as noop in older Arduino Core releases, so let's define it.
        // See also: https://stackoverflow.com/questions/34497758/what-is-the-secret-of-the-arduino-yieldfunction/34498165#34498165
        void yield(void) {};
    #endif
#endif
```

Hope this helps.

Cheers,
Andreas.

bogde, 2019-01-21:
Changes were merged into the main branch.

amotl, 2019-01-21:
Dear Bogdan,

thanks a bunch. However, I really would aim at finally removing the problematic constructor altogether and especially would amend the [ESP8266 example](https://github.com/bogde/HX711/blob/efcf6abb/examples/HX711ESP8266/HX711ESP8266.ino) accordingly.

Would you accept an appropriate pull request?

Thanks,
Andreas.

amotl, 2019-02-17:
Just wanted to let you know we got rid of the `yield()` polyfill altogether [1], so this should never be an issue anymore.

I believe we can well close this issue.

[1] https://github.com/hiveeyes/HX711/tree/spring-cleaning

bogde, 2019-02-21:
Closing this issue. @karesztrk, feel free to open it again if you still have any problems or want to provide any feedback.

closed 2019-02-21 by bogde