bogde/HX711 #120: Replacing yield() by delay(1) | Solution

HamzaHajeir, 2019-01-24:
Hello

I've tested a code in arduino and it works well

But playing with ESP8266 NodeMCU, it comes harder!

[image]

That's because i've commented the while(!isReady()) code in HX711.cpp , because of wdt reset .
then i uncomment that, and replaced yield() by delay(1). it works well.

This is working solution of this issue. Please tell me if there's better one.

Thanks

bogde, 2019-02-21:
hi, @HamzaHajeir! the issue should be fixed in the latest version, thanks to Andreas' effort and the suggestions and contributions made my the community, including yours. i'm going to close this issue but if you have any problems or want to provide any feedback, feel free to open it again at any time. thanks!

closed 2019-02-21 by bogde

HamzaHajeir, 2019-02-21:
Yeah, I followed some similar approach then which is suggested by @el3ctrician, then works well.

Seems now the library deserves another try after the new modifications ! Good job :)