bogde/HX711 #128: New version breaks pre-existing sketches

nseidle, 2019-02-26:
Hi @amotl - I like your expansion of examples, and moving the dat and clk pins to the begin, but your changes to the constructor break years of pre-existing sketches. I recommend you add back init code to the constructor and possibly overload it to support backwards compatibility. See [this issue](https://github.com/sparkfun/OpenScale/issues/29).

@bogde - I'm happy to do a PR but am hesitant as it's not always clear if or when work will get merged. It looks like you're active these days. Are you interested in this?

amotl, 2019-02-26:
Dear @nseidle,

thanks for writing in. I actually thought about doing the facelift in a non-breaking way like some of the people here did by using respective pragma guards. I already reflected this very detail in the [backlog](https://github.com/bogde/HX711/blob/master/doc/notes.md#backlog):
> [o] Maybe use constructor-based initialization again? It does not necessarily need to starting talking to the hardware yet!

However, if we go back to the constructor-as-initializer flavor, we will have to consider some things to make all supported architectures and library maintainers (is it us, actually?) happy **together**. While I will be happy to support you in regaining backward compatibility for this library, I also would like to share some thoughts I learning throughout the process of #123 with both of you.

Cheers,
Andreas.

amotl, 2019-02-27:
## Introduction
We just became aware that while this library was directly referenced by [OpenScale Applications and Hookup Guide » Upgrading the Firmware](https://learn.sparkfun.com/tutorials/openscale-applications-and-hookup-guide#upgrading-the-firmware), the corresponding [OpenScale.ino](https://github.com/sparkfun/OpenScale/blob/master/firmware/OpenScale/OpenScale.ino) would obviously not be in sync with eventual changes coming from upstream.

By introducing the breaking changes, we haven't exactly be aware of the impact this would have to [OpenScale](https://www.sparkfun.com/products/13261) users and want to apologize for the hassle @MIKMAI and you had when being hit with this issue out of the blue after @bogde merged our changes.

## Background
While trying to solve different issues people were having with this library, we came to the conclusion to break the interface intentionally and that updating the [canonical examples](https://github.com/bogde/HX711/tree/master/examples) and documenting the fact by putting a [deprecation warning](https://github.com/bogde/HX711#deprecation-warning) into the README would be enough. From how this has led to problems with downstream users of this library - in this case all users of OpenScale - we are now learning that we should be more careful with things like that in the future.

## Thoughts
Personally, I believe every upstream software artefact should provide stable release packages of its own in order to streamline consumption from downstream. Versioned dependencies you can **actually depend on** are not really a bad thing at all.

After learning bits and pieces of the Arduino library ecosystem when there was no package repository and corresponding library manager out there, we are now happy to see the [PlatformIO Ecosystem](https://platformio.org/) will probably be here to stay and so we all should just start using it to make the magic happen, shouldn't we?

As coming from a Python background, we can tell you the [Python Package Index](https://pypi.org/) is actually one of the best pieces of Python, as the [Comprehensive Perl Archive Network](https://www.cpan.org/) is for Perl.

## Proposed solution
There are currently two ways to provide stable download links referencing release packages of the HX711 library:
- As the library already **is** [available on PlatformIO](https://platformio.org/lib/show/1100/HX711), there's already a direct download link [1].
- When tagging the GitHub repository, GitHub itself will already consider this action as the intention to declare a (stable) release and provide .zip as well as .tar.gz archives of the respective revisions on the [releases page](https://github.com/bogde/HX711/releases) of the repository. This would yield direct download links like [2].

Contents from both archives should yield code which should have been compatible with programs, wrapper libraries and firmwares like OpenScale out there.

[1] http://dl.platformio.org/libraries/archives/11/2527.tar.gz?filename=HX711_0.1_1100
[2] https://github.com/bogde/HX711/archive/v0.1.tar.gz

## Outlook
- We should document the fact that release packages are available through different channels prominently on top of the README page as well as on top of the `HX711.{h,cpp}` files.
- We should actually start tagging releases on the GitHub repository. May we humbly ask @bogde about this? (Remark: We already wanted to open a respective issue ticket, but haven't been able to find additional time for that).

amotl, 2019-02-27:
## Regarding the breaking change itself
> However, if we want to go back to the constructor-as-initializer flavor, we will have to consider some things to make the code run happily on all supported architectures.

We had to decouple the object constructor from calling the `begin()` method implicitly to make the thing work on ESP8266 and ESP32 chips, as with both architectures, Arduino code is not running on the bare metal.

So, when code running on one of these architectures starts talking to GPIO or other hardware interfaces on object initialization time already, it is way too early and things will probably go south.

## Rationale
In order to account for that while not obfuscating the code too much, we had to decide whether to break on compilation time or on runtime and obviously decided for the earlier as the latter would hit users way harder when everything compiles as expected and then would just stop working because nobody did ever call `begin()` before, which is now obligatory, see [earlier version of OpenScale.ino](https://github.com/sparkfun/OpenScale/blob/35872128/firmware/OpenScale/OpenScale.ino).

## Regaining backward compatibility
There actually would be a way to make the library interface backward compatible again. Let's remember we have to a) change the signature of the constructor back to its original shape and b) make the library work even without calling `begin()`.

These steps should be sufficient: 
- Let the constructor obtain the pin definition again.
- Let the library implicitly call `begin()` **once** when being called on `read()`.

## Thoughts
However, is it worth it? If one of you would find regaining backward compatibility important, we will be happy to update the code correspondingly.

bogde, 2019-03-01:
I too apologize for the hassle @nseidle and @MIKMAI had. I was not aware merging the changes would have such an unwanted impact. I do think that these changes add major improvements to the code and solve issues that annoyed the users for too long though. I want to thank @amotl and all the other contributors again for the time and effort they put into this recently.

I just created two releases, one for the code that we had before the spring cleaning and just after changing the license to MIT (0.3.0) and the other one for the current code that breaks compatibility with existing sketches (0.7.1).

I would be happy to pull any changes but, same as Andreas, I wonder if regaining backward compatibility is worth the effort. I'm not saying or implying it isn't, I'm just humbly asking for your opinion.

amotl, 2019-03-01:
Thank you so much, @bodge!

nseidle, 2019-03-01:
No worries folks. Not breaking things can be hard to avoid.

I think I've fixed our firmware and our [hookup guide](https://learn.sparkfun.com/tutorials/load-cell-amplifier-hx711-breakout-hookup-guide/arduino-code) example code so future users should be ok.

I'm ok closing this issue. But I have some recommendations for library encapsulation that I'll start on another issue.