bogde/HX711 #129: Listing this lib on Arduino Library Manager

nseidle, 2019-03-01:
Please package this library for inclusion within the Arduino Library Manager.

* Move cpp and h to \src directory
* Add library.properties file (see [this one](https://github.com/sparkfun/SparkFun_Ublox_Arduino_Library/blob/master/library.properties) as an example)
* Commit everything
* Create release
* Ask Arduino to include your lib by opening [issue on Arduino github](https://github.com/arduino/Arduino/issues)

Background info: This starts from [this discussion](https://github.com/bogde/HX711/issues/128) with @amotl 

> After learning bits and pieces of the Arduino library ecosystem when there was no package repository and corresponding library manager out there, we are now happy to see the [PlatformIO Ecosystem](https://platformio.org/) will probably be here to stay and so we all should just start using it to make the magic happen, shouldn't we?

Funny, I've never used PlatformIO for libs. Have you used the [Arduino Library Manager(https://www.arduino.cc/en/guide/libraries)? I suspect it's how the vast majority of Arduino users consume Arduino libraries. Additionally, I suspect the online Arduino IDE expects libraries so it can consume them, otherwise your lib won't be usable on the Arduino online IDE.

bogde, 2019-03-01:
thanks, @nseidle! i just made the required changes and opened an issue on Arduino github.
@amotl, can you please check if PlatformIO integration still works and give some directions on how to fix that in case it doesn't work anymore. thank you!

nseidle, 2019-03-01:
Looks good! I hope you get it included.

Thanks!

bogde, 2019-03-03:
this is the issue in case anyone wants to show support: https://github.com/arduino/Arduino/issues/8590
(not sure if that matters either)

thanks!

nseidle, 2019-03-03:
They're pretty good about accepting any library as long as it's correctly formed. It can take a few days for a human to get to it, it's not automated as far as I can tell.

bogde, 2019-03-15:
the library has been accepted and it's now available in Arduino Library Manager.

closed 2019-03-15 by bogde

amotl, 2019-03-15:
> the library has been accepted and it's now available in Arduino Library Manager.

Awesome. Thanks @bogde !

nseidle, 2019-03-15:
Saw it the other day in the lib manager. Congrats! Thanks for getting it there. It will help many many people.