bogde/HX711 #130: sketch_mar04a:27:22: error: no matching function for call to 'HX711::HX711(int, int)' / candidate expects 0 arguments, 2 provided

ChrisMLian, 2019-03-04:
Arduino: 1.8.7 (Windows 10), Board: "Arduino/Genuino Uno"

sketch_mar04a:27:22: error: no matching function for call to 'HX711::HX711(int, int)'

 HX711 scale(DOUT, CLK);

                      ^

In file included from G:\Downloads\3d print\Load cell\sketch_mar04a\sketch_mar04a.ino:22:0:

G:\documents\Arduino\libraries\HX711-master\src/HX711.h:30:3: note: candidate: HX711::HX711()

   HX711();

   ^

G:\documents\Arduino\libraries\HX711-master\src/HX711.h:30:3: note:   candidate expects 0 arguments, 2 provided

G:\documents\Arduino\libraries\HX711-master\src/HX711.h:19:7: note: candidate: constexpr HX711::HX711(const HX711&)

 class HX711

       ^

G:\documents\Arduino\libraries\HX711-master\src/HX711.h:19:7: note:   candidate expects 1 argument, 2 provided

exit status 1
no matching function for call to 'HX711::HX711(int, int)'

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

ChrisMLian, 2019-03-04:
nvm solved it by removing the arguments HX711 scale(DOUT, CLK);

Apologies

closed 2019-03-04 by ChrisMLian

griffinlarry95, 2019-03-04:
Found phone around 4 months ago,how do I access it

amotl, 2019-03-04:
Dear Chris,
> nvm solved it by removing the arguments HX711 scale(DOUT, CLK);

Yeah, we changed the signature and the library API recently by #123, which hasn't been made backward compatible on purpose. Sorry that this caused some hassle for you.

The updated examples like [HX711_basic_example.ino](https://github.com/bogde/HX711/blob/0.7.2/examples/HX711_basic_example/HX711_basic_example.ino) show the way it is supposed to work now.

With kind regards,
Andreas.

sluke17, 2020-03-17:
> nvm solved it by removing the arguments HX711 scale(DOUT, CLK);
> 
> Apologies

Any chance you can provide the code to see how you solved this?

amotl, 2020-03-18:
Dear @sluke17,

just have a look at [HX711_basic_example.ino](https://github.com/bogde/HX711/blob/master/examples/HX711_basic_example/HX711_basic_example.ino).

With kind regards,
Andreas.