bogde/HX711 #131: 2's complement and saturation values

MigSanc, 2019-03-06:
Hi all,

I am an electrical engineer student (newbie) trying to implement a circuit using HX711 Board.  During my research I found bogde's code, but since I am trying to implement my own code (to also learn during the process) I am using yours as a guideline to understand how to implement the serial interface communication. However, I have some doubts and I would like to kindly ask if anyone could help me:

1 - During the 24 bit conversion are you/how are you calculating the 2's complement? I saw that you read the 24 bits by shifting one byte at the time and after reading [1],[2],[3],[4],[5] I saw that the conversion is performed in one single line (stated below), but I don't understand how shifting the bytes to a 32 unsigned integer is comparable to performing the 2's complement
  
return (((uint32_t) data[2] << 16) | ((uint32_t) data[1] << 8) | (uint32_t) data[0]);

Can someone give an example or literature to help me understand this?

2- Also, data[2] is compared with 0x80 but 24 bits are not compared with 07FFFFFh (the maximum value as stated in the datasheet). Shouldn't it be? I am just trying to correlate the code with the HX711's datasheet to check if something is missing
[image]

Thanks in advance for your time and for providing the open source code! Great work and thanks for keeping the code updated!

[1] https://github.com/bogde/HX711/issues/23
[2] https://github.com/bogde/HX711/pull/38
[3] https://github.com/bogde/HX711/issues/47

[4] https://github.com/queuetue/Q2-HX711-Arduino-Library/issues/1
[5] https://github.com/queuetue/Q2-HX711-Arduino-Library/pull/5

With best regards,
Miguel

amotl, 2019-03-08:
Dear Miguel,

thanks for writing in. I would also like to point out #110, which is also about to change the way the library reads from the hardware. Maybe it will help for your studies and maybe also for improving the quality of this library.

With kind regards,
Andreas.

MigSanc, 2019-03-11:
Dear Andreas,

Thank you. I will have a look as I am still studying some things related to the HX711 code.

With best regards,
Miguel

jgav1, 2019-11-05:
@MigSanc  did you managed to get the 2's complement ? I would also like the HX711 to read +- Voltage

MigSanc, 2019-11-05:
Hi @jgav1 

If you check bodge library they just shift the 24 bits (pulses) of data to construct a 32-bit integer and then cast into long type to return it. So, you now know 2 things:
1 - I know that I am reading a 24 bit word 
2 - the range of signed integers that can be represented in 24 bits is −8,388,608 to 8,388,607

So, I never cared with two complement's because I know the range.
Concerning the voltage is simple math. Check the HX711 datasheet as well, it is important.

Just think now with me:
The HX711 is a differential amplifier with a Programable Gain (PGA).
So, the differential input signal is being amplified and then converted by the internal ADC to a 24-bit word within the range, right?
What happens if the differential input signal is 0? (what will be the expected ADC value?)
What happens if you have the maximum differential input signal? 

I will point you in the right direction once you answer me to these questions :) I want you to think as well on this matter. Sorry for the long explanation.

Best regards

jgav1, 2019-11-06:
Hi Miguel, im still thinking about your hints, but I haven't figured
anything yet.  I would like to know if you where you able to get negative
values ? ( 0X800000 to 0XFFFFFF)
When I use the library and connect the channel A with +4.5 mv I get an
output from the Hx711 however when I apply the same voltage but reversed,
that is -4.5 MV I don't get an stable nor the same value but inversed.
For example:
If I connect 4.5 mv I get around 5 M in decimal and with -4.5 mv I get
around -320k in decimal.

According to the datasheet the chip should be able to read both positive
and negative volatages. Since I'm not using it just for compression, but
also for tension, I need to be able to read positive as well as negative
values.
Thanks in advance

-- 
AVISO DE CONFIDENCIALIDAD. El contenido de este correo electrónico, 
incluyendo cualquier documento adjunto es confidencial y puede contener 
información privilegiada propiedad de la Universidad de Monterrey y/o 
entidades relacionadas. Se entiende dirigido y para uso exclusivo del 
destinatario, si usted ha recibido este mensaje por error o no le compete, 
está prohibida su difusión, transmisión, alteración, explotación, impresión 
o copia. Por favor inmediatamente avise al remitente y elimine el original. 
Para mayor información sobre nuestro Aviso de confidencialidad y exención - 
Confidential notice and legal advice disclaimer consulte:  
www.udem.edu.mx/disclaimer 
<https://admin.google.com/www.udem.edu.mx/disclaimer>

MigSanc, 2019-11-06:
Hi @jgav1 

Do you have a circuit diagram so that I can understand your connections?
Please provide as much info as necessary.

Please try something for me first. I believe you are connecting the 4.5mV to channel A input + and GND for the channel A input -, correct? 
and how are you connecting the -4.5mV? to the channel A input +? and the input - to GND?

I was able to read negative values. Send me a connection diagram to check both cases. Remember that you are reading a differential input, so if you want to read -4.5mV, you should connect the 4.5mV (positive) to input A negative and GND to input A positive. Test it out and let me know if it worked. 
I believe that your mistake is on trying reading a voltage value that is lower than the low rail of the amplifier of the HX711, which is GND, right? ;)

jgav1, 2019-11-06:
Hi @MigSanc 

Sure I will send you a simplified schematic of what I mean.  The resistors in the schematic are Strain gauges for a custom made transducer. the thing is that the resistance and hence the voltage  vary, so the voltage accross A+ A- can either be positive or negative ( in this particualr schematic +-4.5mv)

According to the following image, the voltage across terminals is 4.5mv ( which is a situation that Im currently experiencing ) 

the voltage across R2 and E- is 2500mv. the voltage across R4 and E- is 2495.5 mv so the expected voltage across A+, A- is 4.5mv
[image]

The next image shows the situation at which the strain gauges change polarity

the voltage across R2 and E- is 2495.5 mv. the voltage across R4 and E- is 2500 mv so the expected voltage across A+, A- is -4.5mv

[image]

The problem is that my HX711 ( using the current library ) can register the 4.5mv, but it doesnt reads the -4.5mv. 

Where you able to read negative voltages according to  what I meant?

I dont believe the voltage is lower than the low rail suppliy as you state and according to the Schematic of the HX711 it should be able to register negative voltages like this. 

Thanks in advance
Jorge Aragon

Ps: is there a contact email to where I can contact you quicker ?

MigSanc, 2019-11-06:
Hi Jorge, 

tell me your email and I will reply from now on.
From your schematics, everything seems fine. 
And I was able to read from a Wheatstone bridge of strain gauges as well.

Best regards,
Miguel

MigSanc, 2019-11-06:
@jgav1 

What code are you using?
What gain did you set?
What are the pins from the UNO that you have connected to DOUT and SCK pins from the HX711 module?
Are you connecting to the 5V pin from the arduino uno?

jgav1, 2019-11-06:
jorge.aragon@udem.edu

-- 
AVISO DE CONFIDENCIALIDAD. El contenido de este correo electrónico, 
incluyendo cualquier documento adjunto es confidencial y puede contener 
información privilegiada propiedad de la Universidad de Monterrey y/o 
entidades relacionadas. Se entiende dirigido y para uso exclusivo del 
destinatario, si usted ha recibido este mensaje por error o no le compete, 
está prohibida su difusión, transmisión, alteración, explotación, impresión 
o copia. Por favor inmediatamente avise al remitente y elimine el original. 
Para mayor información sobre nuestro Aviso de confidencialidad y exención - 
Confidential notice and legal advice disclaimer consulte:  
www.udem.edu.mx/disclaimer 
<https://admin.google.com/www.udem.edu.mx/disclaimer>

GeekyDeaks, 2019-12-28:
I think there is a problem with the 2's complement logic.  The filler byte should go between `data[2]` and `data[1]`, but it's actually getting shifted all the way to the MSB

https://github.com/bogde/HX711/blob/451eb93ebe149a326eb0666b114c2c733b770076/src/HX711.cpp#L171-L182

jgav1, 2019-12-28:
Thank you ! I already solved it, I will upload the code so that everyone
can use it.
With the modifications now is possible to  measure tension and compression

-- 
AVISO DE CONFIDENCIALIDAD. El contenido de este correo electrónico, 
incluyendo cualquier documento adjunto es confidencial y puede contener 
información privilegiada propiedad de la Universidad de Monterrey y/o 
entidades relacionadas. Se entiende dirigido y para uso exclusivo del 
destinatario, si usted ha recibido este mensaje por error o no le compete, 
está prohibida su difusión, transmisión, alteración, explotación, impresión 
o copia. Por favor inmediatamente avise al remitente y elimine el original. 
Para mayor información sobre nuestro Aviso de confidencialidad y exención - 
Confidential notice and legal advice disclaimer consulte:  
www.udem.edu.mx/disclaimer 
<https://admin.google.com/www.udem.edu.mx/disclaimer>

Trishaprophecy, 2022-05-17:
Hi, everyone I am using HX711 ADC and interface it with Stm32. I am supplying 5v in the HX711 adc and in channel A, applying 16.8mv but getting 0v as output voltage or 3.3 v, ADC is getting saturated , resolving this I am applying zero voltage at the differential in of channel (A+, A-), unplugging the stm32L4766RG module .still saturation problem remains same. 
Could anyone help me ? the raw data is 2's complemented in the code. I am giving 25 pulse by using PDSCK pin.

Thanks and regards.
Trisha Kundu