bogde/HX711 #132: NodeMCU + HX711 + 4 50kgs Load Cells Returns "-83388608"

TommiPio, 2019-03-08:
Hello, I'm trying to use 4 50kgs loadcells with a nodemcu and an hx711 to create a Bed Presence Sensor.

[This is my circuit](https://selfhostedhome.com/wp-content/uploads/2018/09/schematic.png)

What happen, is that the serial monitor just reports "-83388608" or "83388607" as value. I'm using the hx711 library by Bogde, but I also tried other two different hx711 libraries and all of them gave me the same result. 

Anyway, this is the code I'm using now.

<pre>
<font color="#5e6d03">#include</font> <font color="#434f54">&lt;</font><font color="#000000">Arduino</font><font color="#434f54">.</font><font color="#000000">h</font><font color="#434f54">&gt;</font>
<font color="#5e6d03">#include</font> <font color="#434f54">&lt;</font><b><font color="#d35400">ArduinoOTA</font></b><font color="#434f54">.</font><font color="#000000">h</font><font color="#434f54">&gt;</font>
<font color="#5e6d03">#include</font> <font color="#434f54">&lt;</font><font color="#d35400">ESP8266WiFi</font><font color="#434f54">.</font><font color="#000000">h</font><font color="#434f54">&gt;</font>
<font color="#5e6d03">#include</font> <font color="#434f54">&lt;</font><b><font color="#d35400">PubSubClient</font></b><font color="#434f54">.</font><font color="#000000">h</font><font color="#434f54">&gt;</font>

<font color="#5e6d03">#include</font> <font color="#005c5f">&#34;config.h&#34;</font>
<font color="#5e6d03">#include</font> <font color="#005c5f">&#34;HX711.h&#34;</font>

<font color="#d35400">WiFiClient</font> <font color="#000000">espclient</font><font color="#000000">;</font>
<b><font color="#d35400">PubSubClient</font></b> <font color="#d35400">client</font><font color="#000000">(</font><font color="#000000">espclient</font><font color="#000000">)</font><font color="#000000">;</font>

<b><font color="#d35400">HX711</font></b> <font color="#000000">scale</font><font color="#000000">;</font>

<font color="#00979c">void</font> <font color="#d35400">reconnect</font><font color="#000000">(</font><font color="#000000">)</font> <font color="#000000">{</font>
 &nbsp;&nbsp;&nbsp;<font color="#434f54">&#47;&#47; Loop until we&#39;re reconnected</font>
 &nbsp;&nbsp;&nbsp;<font color="#5e6d03">while</font> <font color="#000000">(</font><font color="#434f54">!</font><font color="#d35400">client</font><font color="#434f54">.</font><font color="#d35400">connected</font><font color="#000000">(</font><font color="#000000">)</font><font color="#000000">)</font> <font color="#000000">{</font>
 &nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;<b><font color="#d35400">Serial</font></b><font color="#434f54">.</font><font color="#d35400">print</font><font color="#000000">(</font><font color="#005c5f">&#34;Attempting MQTT connection...&#34;</font><font color="#000000">)</font><font color="#000000">;</font>
 &nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;<font color="#434f54">&#47;&#47; Attempt to connect</font>
 &nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;<font color="#5e6d03">if</font> <font color="#000000">(</font><font color="#d35400">client</font><font color="#434f54">.</font><font color="#d35400">connect</font><font color="#000000">(</font><font color="#000000">HOSTNAME</font><font color="#434f54">,</font> <font color="#000000">MQTT_USER</font><font color="#434f54">,</font> <font color="#000000">MQTT_PASSWORD</font><font color="#000000">)</font><font color="#000000">)</font> <font color="#000000">{</font>
 &nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;<b><font color="#d35400">Serial</font></b><font color="#434f54">.</font><font color="#d35400">println</font><font color="#000000">(</font><font color="#005c5f">&#34;connected&#34;</font><font color="#000000">)</font><font color="#000000">;</font>
 &nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;<font color="#434f54">&#47;&#47; Once connected, publish an announcement...</font>
 &nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;<font color="#d35400">client</font><font color="#434f54">.</font><font color="#d35400">publish</font><font color="#000000">(</font><font color="#000000">AVAILABILITY_TOPIC</font><font color="#434f54">,</font> <font color="#005c5f">&#34;online&#34;</font><font color="#000000">)</font><font color="#000000">;</font>
 &nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;<font color="#000000">}</font> <font color="#5e6d03">else</font> <font color="#000000">{</font>
 &nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;<b><font color="#d35400">Serial</font></b><font color="#434f54">.</font><font color="#d35400">print</font><font color="#000000">(</font><font color="#005c5f">&#34;failed, rc=&#34;</font><font color="#000000">)</font><font color="#000000">;</font>
 &nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;<b><font color="#d35400">Serial</font></b><font color="#434f54">.</font><font color="#d35400">print</font><font color="#000000">(</font><font color="#d35400">client</font><font color="#434f54">.</font><font color="#000000">state</font><font color="#000000">(</font><font color="#000000">)</font><font color="#000000">)</font><font color="#000000">;</font>
 &nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;<b><font color="#d35400">Serial</font></b><font color="#434f54">.</font><font color="#d35400">println</font><font color="#000000">(</font><font color="#005c5f">&#34; try again in 5 seconds&#34;</font><font color="#000000">)</font><font color="#000000">;</font>
 &nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;<font color="#434f54">&#47;&#47; Wait 5 seconds before retrying</font>
 &nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;<font color="#d35400">delay</font><font color="#000000">(</font><font color="#000000">5000</font><font color="#000000">)</font><font color="#000000">;</font>
 &nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;<font color="#000000">}</font>
 &nbsp;&nbsp;&nbsp;<font color="#000000">}</font>
<font color="#000000">}</font>

<font color="#00979c">void</font> <font color="#5e6d03">setup</font><font color="#000000">(</font><font color="#000000">)</font> <font color="#000000">{</font>

 &nbsp;&nbsp;&nbsp;
 &nbsp;&nbsp;&nbsp;<b><font color="#d35400">Serial</font></b><font color="#434f54">.</font><font color="#d35400">begin</font><font color="#000000">(</font><font color="#000000">115200</font><font color="#000000">)</font><font color="#000000">;</font>
 &nbsp;&nbsp;&nbsp;<b><font color="#d35400">Serial</font></b><font color="#434f54">.</font><font color="#d35400">println</font><font color="#000000">(</font><font color="#000000">)</font><font color="#000000">;</font>
 &nbsp;&nbsp;&nbsp;<font color="#000000">scale</font><font color="#434f54">.</font><font color="#d35400">begin</font><font color="#000000">(</font><font color="#000000">5</font><font color="#434f54">,</font> <font color="#000000">4</font><font color="#000000">)</font><font color="#000000">;</font>

 &nbsp;&nbsp;&nbsp;<b><font color="#d35400">WiFi</font></b><font color="#434f54">.</font><font color="#d35400">mode</font><font color="#000000">(</font><font color="#00979c">WIFI_STA</font><font color="#000000">)</font><font color="#000000">;</font>
 &nbsp;&nbsp;&nbsp;<b><font color="#d35400">WiFi</font></b><font color="#434f54">.</font><font color="#d35400">begin</font><font color="#000000">(</font><font color="#d35400">SSID</font><font color="#434f54">,</font> <font color="#000000">PASSWORD</font><font color="#000000">)</font><font color="#000000">;</font>
 &nbsp;&nbsp;&nbsp;<b><font color="#d35400">Serial</font></b><font color="#434f54">.</font><font color="#d35400">print</font><font color="#000000">(</font><font color="#005c5f">&#34;Connecting...&#34;</font><font color="#000000">)</font><font color="#000000">;</font>

 &nbsp;&nbsp;&nbsp;<font color="#5e6d03">while</font> <font color="#000000">(</font><b><font color="#d35400">WiFi</font></b><font color="#434f54">.</font><font color="#d35400">status</font><font color="#000000">(</font><font color="#000000">)</font> <font color="#434f54">!=</font> <font color="#000000">WL_CONNECTED</font><font color="#000000">)</font> <font color="#000000">{</font>
 &nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;<font color="#d35400">delay</font><font color="#000000">(</font><font color="#000000">500</font><font color="#000000">)</font><font color="#000000">;</font>
[... cut ...]
 &nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;<font color="#000000">lastValue</font> <font color="#434f54">=</font> <font color="#000000">value</font><font color="#000000">;</font>
 &nbsp;&nbsp;&nbsp;<font color="#000000">}</font>

<font color="#000000">}</font>

</pre>

I also have two copies of this sensor, one soldered to a protoboard and one tested using a breadboard, both seems to give me the same problem.

What can I do? 

The wifi and mqtt parts are working fine. The value output from the hx711 is "-8388608" when vcc connected to 3v pin or VU pin (that should output 5v). 

I'm quite a noob, but if you guys could give me some advice I would love it.

amotl, 2019-03-08:
Dear @TommiPio,

[`read_average()` says it would return a `long`](https://github.com/bogde/HX711/blob/0.7.2/src/HX711.h#L59-L60), but your `value` for storing the outcome is defined as `int`. Maybe a type cast going wrong yields the weird reading value you are seeing?

With kind regards,
Andreas.

TommiPio, 2019-03-08:
Thanks Andreas, unfortunately this didn't solve it.

I kinda made it work, but in a strange way, I really need someone to explain me what is going on with this.

Normally, as seen in my schematic, the VCC pin on the hx711 should be connected to the 3v pin, but this stuck the output to the infamous value, -8388608.

I was playing with my cables, and I noticed that if I link the 3v pin of nodemcu, to the E+ pin on the HX711, suddenly the load cells are working fine.

What am I missing here? Could I actually wire my sensor this way without even passing from the vcc pin on hx711?

> 16:44:17.150 -> Connecting....
> 16:44:17.632 -> Connected, IP address: 192.168.1.136
> 16:44:17.632 -> Attempting MQTT connection...connected
> 16:44:19.946 -> -1      -------------> 3v pin not connected to anything
> 16:44:21.216 -> -1       
> 16:44:22.451 -> -1
> 16:44:35.996 -> -8388608            -------------->3v pin connected to vcc, as in the diagram
> 16:44:38.131 -> -8388608
> 16:44:43.015 -> -8388608
> 16:44:45.150 -> -8388608
> 16:44:47.283 -> -8388608
> 16:44:48.525 -> -1               ------------->3v pin disconnected again
> 16:44:50.589 -> -327076    ------------> 3v pin connected to E+ pin (vcc not connected to anything)
> 16:44:52.714 -> -326920
> 16:44:54.885 -> -202974 ------>applying load to the loadcells, it works
> 16:44:57.010 -> -223136
> 16:44:59.142 -> -194757
> 16:45:01.278 -> -165867
> 16:45:03.413 -> -362636
> 16:45:05.581 -> -329727
> 16:45:07.712 -> -327398
> 16:45:09.850 -> -327435

bogde, 2019-03-08:
Are you sure your HX711 board is good? Do you have another one to test?

TommiPio, 2019-03-08:
Hi @bogde, thanks for this library. I already tried on a breadboard with new hx711, loadcells, and nodemcu, same problem applied.

Everything came from the same reseller, so maybe both hx711 were faulty (?)

Anyway, I can confirm that at least with one of the hx711, [this wiring is working](https://imgur.com/a/U4cb6KD). Do I risk of burning something with such a wiring? Does it have any sense that it's working like this for you?

ClemensGruber, 2019-03-08:
@TommiPio two things: doublecheck your wiring. You are using a half bridge setting, so perhaps the schematic of the Sparkfun load cell combinator can help: https://github.com/sparkfun/Load_Sensor_Combinator/blob/V_1.1/Hardware/SparkFun%20Load%20Sensor%20Combinator.sch Second: did you "calibrate" (more correct ist "justify") your setting and tare your setting? You have to do this before. Are this values in the config file already? Seems that you are reading the raw values only and no kg values, see the example https://github.com/bogde/HX711/blob/master/examples/HX711_full_example/HX711_full_example.ino

isnainiahmadi, 2019-03-12:
I'm pretty sure that you use separated power supply (not directly from Arduino). If you do, connect power supply ground to Arduino too.

piotrbazan, 2019-04-03:
Had same problem here with the same strange value. Spent several hours until I noticed that manufacturer did not conform to standard of sensor color cables. In my case conneting Black Power + , white Power - Red Signal + green signal - solved the issue. Hope it helps.

amotl, 2019-10-17:
Things like constant readings of `INT24_MAX` (`8388607L`) might also be caused by faulty hardware. Enjoy reading [1] just reported by @poesel. Thanks!

[1] https://community.hiveeyes.org/t/hx711-with-arduino-uno-constant-8388607-output/2603/3

prof-kalabwa, 2020-01-04:
```
#include <Arduino.h>
#include <ArduinoOTA.h>
#include <ESP8266WiFi.h>
#include <PubSubClient.h>

#include "config.h"
#include "HX711.h"

WiFiClient espclient;
PubSubClient client(espclient);

HX711 scale;

void reconnect() {
    // Loop until we're reconnected
    while (!client.connected()) {
        Serial.print("Attempting MQTT connection...");
        // Attempt to connect
        if (client.connect(**HOSTNAME**, **MQTT_USER**, **MQTT_PASSWORD**)) {
            Serial.println("connected");
            // Once connected, publish an announcement...
            client.publish(**AVAILABILITY_TOPIC**, "online");
        } else {
            Serial.print("failed, rc=");
            Serial.print(client.state());
            Serial.println(" try again in 5 seconds");
            // Wait 5 seconds before retrying
            delay(5000);
        }
    }
}

void setup() {

    
    Serial.begin(115200);
    Serial.println();
    scale.begin(5, 4);

    WiFi.mode(WIFI_STA);
    WiFi.begin(**SSID**, **PASSWORD**);
    Serial.print("Connecting...");

    while (WiFi.status() != WL_CONNECTED) {
        delay(500);
        Serial.print(".");
    }
    Serial.println();

    Serial.print("Connected, IP address: ");
    Serial.println(WiFi.localIP());

    client.setServer(**MQTT_SERVER**, 1883);
}

void loop() {
  static int lastValue = 0;
    int value;
  if (!client.connected()) {
      reconnect();
   
    }
   value = scale.read_average(5);
    Serial.println(value);

    // Only publish new value if difference of 500 or more
    if (abs(lastValue - value) >= 500) {
        String value_str = String(value);
        client.publish(STATE_TOPIC, (char *)value_str.c_str());
        lastValue = value;
    }

}
```
 
i've got errors about how the highlighted variables are not declared and I don't know what to declare on that variables. please tell me how to declare that variables or reference for me to read about that variables. thank you

ClemensGruber, 2020-01-04:
@prof-kalabwa where is your code from? You may have a look at the example there. It seems that `config.h `is the file you should check. And you may have a look athttps://github.com/knolleary/pubsubclient/blob/master/examples/mqtt_esp8266/mqtt_esp8266.ino

HamzaHajeir, 2020-06-07:
> Had same problem here with the same strange value. Spent several hours until I noticed that manufacturer did not conform to standard of sensor color cables. In my case conneting Black Power + , white Power - Red Signal + green signal - solved the issue. Hope it helps.

Same here, I lastly went to the selling page that describes this info, changed and fixed.

rn-ksheer, 2021-01-28:
> Thanks @amotl , unfortunately this didn't solve it.
> 
> I kinda made it work, but in a strange way, I really need someone to explain me what is going on with this.
> 
> Normally, as seen in my schematic, the VCC pin on the hx711 should be connected to the 3v pin, but this stuck the output to the infamous value, -8388608.
> 
> I was playing with my cables, and I noticed that if I link the 3v pin of nodemcu, to the E+ pin on the HX711, suddenly the load cells are working fine.
> 
> What am I missing here? Could I actually wire my sensor this way without even passing from the vcc pin on hx711?
> 
> > 16:44:17.150 -> Connecting....
> > 16:44:17.632 -> Connected, IP address: 192.168.1.136
> > 16:44:17.632 -> Attempting MQTT connection...connected
> > 16:44:19.946 -> -1      -------------> 3v pin not connected to anything
> > 16:44:21.216 -> -1
> > 16:44:22.451 -> -1
> > 16:44:35.996 -> -8388608            -------------->3v pin connected to vcc, as in the diagram
> > 16:44:38.131 -> -8388608
> > 16:44:43.015 -> -8388608
> > 16:44:45.150 -> -8388608
> > 16:44:47.283 -> -8388608
> > 16:44:48.525 -> -1               ------------->3v pin disconnected again
> > 16:44:50.589 -> -327076    ------------> 3v pin connected to E+ pin (vcc not connected to anything)
> > 16:44:52.714 -> -326920
> > 16:44:54.885 -> -202974 ------>applying load to the loadcells, it works
> > 16:44:57.010 -> -223136
> > 16:44:59.142 -> -194757
> > 16:45:01.278 -> -165867
> > 16:45:03.413 -> -362636
> > 16:45:05.581 -> -329727
> > 16:45:07.712 -> -327398
> > 16:45:09.850 -> -327435

Excellent mod !!!!, After a day of search i was able to get loadcell to work. Leaving 5v connected to VCC on HX711 module or disconnecting it did not many any difference.

TommiPio, 2021-01-28:
I didn't end up using this wiring in the end because the value were not stable enough for me. I really think it depends on the Hx711 unit, I read that with the sparkfun one the problems are almost non-existent.

Anyway, I always dreamt about understanding why this nonsense wiring schemed I found out seemed to be working ahaha

rn-ksheer, 2021-01-28:
I wanted to detect weight changes and trigger a action so value's accuracy was not a concern.