bogde/HX711 #136: Trying to read values during an Interruption service routine

Luigi94100, 2019-03-25:
Hi,

I tried to use the scale.get_units() function in an ISR.
The same function works perfectly in the void loop() but in the ISR, I get only one value and then nothing.

I manage to read other sensors values in the ISR by declaring volatile variables.
These sensors do not use any library.
I suppose I have to modify the library but I don't know how.

I could use some help.

Best regards,

amotl, 2019-03-25:
Dear Luigi,

thanks for writing in.

`get_units()` calls `read()` under the hood, which in turn might go into a spin lock waiting for the device to be ready. I believe the current architecture regarding to this is just not ready for interrupt-based reading and would have to be amended appropriately.

May we humbly ask about which architecture you are targeting? We recently found an interrupt-based implementation for ESP32 at [1] and reported about it at [2], which might help. Please let us know about your progress or outcome and feel free to share further thoughts about this topic.

With kind regards,
Andreas.

[1] https://github.com/espressif/arduino-esp32/issues/1289#issuecomment-379041436
[2] https://github.com/bogde/HX711/issues/125#issuecomment-475874114

mcd8604, 2019-05-17:
I'm using an Uno, but I've managed to use an ISR on the DOUT pin. This effectively avoids the blocking loop (in wait_ready()) because the DOUT pin should always be LOW when read() is called. Do note that I am passing 1 to get_units which will only shiftIn the serial data from the HX711 a single time. The call to read() will cause the DOUT pin to return HIGH. If you pass a value greater than 1, then every subsequent read() call will block in wait_ready().

I haven't tried this on a Due, so please let me know if you try this.

Here are the steps I took:
- Add flag to sketch:
`volatile bool hx711Ready = false;`
- Attach interrupt in sketch (in setup) which simply sets the flag:
`attachInterrupt(digitalPinToInterrupt(LOADCELL_DOUT_PIN), []{ hx711Ready=true; }, FALLING);`
- Check the flag in the main loop of the sketch and call read:
```
    if(hx711Ready) {
        currentWeight = loadcell.get_units(1);
        Serial.println(currentWeight);
        hx711Ready = false;
    }
```

ericnickel69, 2021-01-31:
Did you ever find a solution for this?  I'm trying to do the same thing but not having any luck getting good data.

Thanks,
Eric