bogde/HX711 #140: build-all doesn't build-anything

csoeder, 2019-04-12:
I tried to get this library working on a new computer and ran into trouble following the README. Cloning the directory works fine
_$ git clone https://github.com/bogde/HX711
$ cd HX711/_
Then, following the README, the build starts off running the virtual environment without trouble but ultimately fails:
_$ make build-all 
Running virtualenv with interpreter /usr/bin/python2
... Installing setuptools, pkg_resources, pip, wheel...done.
...
============================================================= [SUMMARY] =============================================================
Environment feather_328 [ERROR]
Environment atmega_2560 [ERROR]
Environment huzzah      [ERROR]
Environment lopy4       [ERROR]
Environment teensy31    [ERROR]
Environment teensy36    [ERROR]
Environment feather_m0  [ERROR]
Environment arduino_due [ERROR]
Environment feather_m4  [ERROR]
Environment bluepill    [ERROR]
==================================================== [ERROR] Took 246.14 seconds ====================================================
make: *** [Makefile:28: build-all] Error 1_

I can include the whole output, but the key error message seems to be:

_path/HX711/examples/HX711_basic_example/HX711_basic_example.ino:1:19: fatal error: HX711.h: No such file or directory

***************************************************************
* Looking for HX711.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:HX711.h"
* Web  > https://platformio.org/lib/search?query=header:HX711.h
*
***************************************************************

compilation terminated.
*** [.pioenvs/feather_328/src/HX711_basic_example.ino.cpp.o] Error 1
===================================================== [ERROR] Took 5.11 seconds =====================================================_

Indeed, there is no HX711.h or :HX711.cpp file in the cloned directory HX711/ , but there are these files in HX711/src/. When copied to the upper HX711/ folder, 
_$ make build-all 
...
============================================================= [SUMMARY] =============================================================
Environment feather_328 [SUCCESS]
Environment atmega_2560 [SUCCESS]
Environment huzzah      [SUCCESS]
Environment lopy4       [SUCCESS]
Environment teensy31    [SUCCESS]
Environment teensy36    [SUCCESS]
Environment feather_m0  [SUCCESS]
Environment arduino_due [SUCCESS]
Environment feather_m4  [SUCCESS]
Environment bluepill    [SUCCESS]
=================================================== [SUCCESS] Took 220.75 seconds ===================================================_

The example scripts for this installation compile without error.

I'm not fluent enough with make to pinpoint the problem, but it looks like the HX711.h and HX711.cpp files are not where the installation script is expecting them to be.

epatel, 2020-01-01:
I had the same initial problem. But poking around I got it to work by changing `platformio.ini` a little.

I changed `include` from `.` to `src`
and `src_filter` part `+<../../*.cpp>` to `+<../../src/*.cpp>`

amotl, 2020-01-01:
Dear @epatel,

thanks a bunch. Could you make a PR out of your findings?

With kind regards,
Andreas.

amotl, 2020-01-03:
Thanks for #169, @epatel! We might want to close this issue now.

closed 2021-03-04 by bogde