bogde/HX711 #146: Building on Wondows 10

RandoSY, 2019-07-13:
Wow, what a great library...

I was interested in your build setup as it seems very modern and uses Platform.io, and was wondering if there was any special setup I needed to do before I used it?

I have both Python2, and 3 installed on my machine, and the Python 2 is in the path, but I make sure to rename the Python directory to Python2_Disarm, so it will go to Python 3 by default.

Also I note that maybe the makefile is expecting to be run from Linux? 

Anyway, any guidance much appreciated!

amotl, 2019-07-13:
Dear @RandoSY,

while we haven't heard about specific things to care for about Windows, some parts of the build infrastructure might well be suited for Linux only. However, we've had very good experiences with running anything Linux-related almost without any efforts under WSL2, the revamped and improved Windows Subsystem for Linux in general. If you have more specific questions, @ClemensGruber might be able to assist.

In general, you might be fine installing this library through the Arduino Library Manager if you are programming directly within this environment. In any case, a recent `avr-gcc` should do or even other compatible build chains like `xtensa-gcc`. `platformio` just gives assistance in getting everything on the plate conveniently. As we are not using this that often, we currently can't tell things about Windows support for `platformio` - but we assume it should be there already?

With kind regards,
Andreas.

bogde, 2021-03-04:
On Windows 10, I found that using WSL works great and it's easy enough for this and other scenarios that require gcc or other build tools.

I will try to write some directions to help interested people in getting started with this.

First enable Windows Subsystem for Linux by using "Turn Windows features on or off". Open "Turn Windows features on or off" (you can just type "turn windows ..." in the Windows search bar), select "Windows Subsystem for Linux", click OK and reboot.

Then, open the Microsoft Store app, install Ubuntu and launch it.

In Ubuntu, you can access your Windows drives under ```/mnt/```, like this:
```
cd /mnt/c/projects/hx711
```

Your path will most likely be different; the letter ```c``` in the above example designates your C: drive.

In the HX711 folder just run:
```
make clean
make build-all
```

I hope this helps.