bogde/HX711 #147: How can I get the old library?

DVB-Arduino, 2019-07-15:
How can I get the old library? My code works great on the old library. But after updating the library I need to update all my code.

amotl, 2019-07-15:
Dear @DVB-Arduino,

we really encourage you to use the overhauled library directly from the Arduino Library Manager. Which kind of obstacles are you experiencing with your old code? It should be trivial to update it.

Otherwise, you might want to look at [1].

With kind regards,
Andreas.

[1] https://github.com/bogde/HX711/tree/0.3.0

DVB-Arduino, 2019-08-05:
OK. Thank. I rewrote my code to a new library. I get the values in the format 1938538.00. How do I convert this to kilograms?

bogde, 2019-08-05:
@DVB-Arduino You may want to take a look at HX711_full_example.ino

cheers!

bogde, 2021-03-04:
Closing due to lack of feedback.

closed 2021-03-04 by bogde