bogde/HX711 #148: Improvement of reading ADC values from HX711

vrbadev, 2019-07-18:
Hello,
I think that better implementation than current could be presented. The final scale value from averaged 24-bit signed integers doesn't make sense because of not considered ENOB (effective number of bits) of the 24-bit ADC.
From IEEE standard, the formula for the ENOB value is:
![Formula](https://latex.codecogs.com/png.latex?%5Ctext%7BENOB%7D%20%3D%20%5Clog_%7B2%7D%20%5Cleft%20%28%20%5Cfrac%7BV_%7B%5Ctext%7Bin%20full-scale%7D%7D%7D%7BV_%7B%5Ctext%7Bnoise%7D%7D%7D%20%5Cright%20%29)
From the datasheet of HX711, the full-scale range is 
![Full-scale range](https://latex.codecogs.com/png.latex?%5Cpm%200.5%5Cleft%20%28%20%5Ctext%7BAVDD%7D/%5Ctext%7BGAIN%7D%20%5Cright%20%29)
and the noise (specified for gain of 128 only) is 50 nV (on the HX711, gain pin is grounded, thus sample rate is 10 SPS).
The final ENOB value is 17.78 bits (for gain 128).
So I think the 7 least significant bits could be always truncated from the calculation, as they have no meaning (affected by the noise). Without removing them the averaged result is affected by the noise, hence not correct.
This is just a proposal of improving the code.

scotabbott, 2021-04-13:
I found essentially the same thing using a common signal to noise calculation. bits = log base 2 (full scale signal/ stdev( tared value)

amotl, 2022-06-02:
Dear Bogdan,

regarding your question about this topic over at https://github.com/bogde/HX711/issues/172#issuecomment-790589521, where you asked:

> I know it's been some time since that was posted but i'd appreciate your thoughts on it. I've been doing tests lately and, as expected, that does make readings more stable. I was thinking to add some new methods that would allow the user to basically configure the library to set a number of bits to `0` in the `read()` function. Something like:
>
> `data[0] = data[0] & 128`

I have to admit that I am not too much of a hardware-versed programmer who could add anything sensible to this discussion. Apologies. Maybe @msweef is able to contribute some thoughts on this topic.

On the other hand, if you want to have any kinds implementation improvements being tested by people from our community at [Hiveeyes](https://hiveeyes.org/), I will be happy to lend a hand and carry the discussion back and forth on [our forum](https://community.hiveeyes.org/). Don't hesitate to ping me about it.

With kind regards,
Andreas.