bogde/HX711 #150: Before Calibration already getting 2 Kg fluctuation - Is that normal ?

mggil7, 2019-07-23:
Hello ,

I am facing problems and I wonder if someone is willing to help me out.

The scenario is this?

ESP32 WROOM 32
4 – 50 kg load cells
01 – HX711
Library : https://github.com/bogde/HX711/
SERIAL OUTPUT is below
Code is below
As serial output, I am getting weird values when I start to calibrate my load cells.
Before calibration, the weight amplitude is around 2 Kg and I was hoping to get less 100 grams .
TWO MAIN QUESTIONS :

FIRST – How can I minimize the weight out fluctuation in order to reach 100 grams or less ?
 SECOND – What are these “crazy” symbols at the beginning of serial output ?

What am I missing here? Please advise.

Any suggestion, help, previous experience is more than welcome.

Thank you. 

SERIAL OUTPUT
Windows PowerShell
Copyright (C) 2009 Microsoft Corporation. Todos os direitos reservados.

PS C:\Users\RITA\Documents\PlatformIO\Projects\Exemplo HX711> pio device monitor --port COM13
--- Miniterm on COM13  9600,8,N,1 ---
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
���␚�   z�␔␁␌   ␑␐␁␁␐␄␀␔P␀ ␁␁␁␐A␁%␓�␁␁␂I␑␀@␀` �␒�L�>␅R␄␐�y�VՔ+8I␝�HX711 Demo
Before setting up the scale:
read:           789
read average:           789
get value:              776.00
get units:              793.0
After setting up the scale:
read:           792
read average:           822

get value:              -16.00
get units:              0.0
Readings:
one reading:    1.2     | average:      0.1
one reading:    1.6     | average:      0.6
one reading:    0.2     | average:      0.8
one reading:    0.4     | average:      0.9
one reading:    2.3     | average:      1.7

CODE

#include <Arduino.h>
#include "HX711.h"
#include "soc/rtc.h"

// HX711.DOUT	- pin #A1
// HX711.PD_SCK	- pin #A0

HX711 scale(17, 16);		// parameter "gain" is ommited; the default value 128 is used by the library

void setup() {
  rtc_clk_cpu_freq_set(RTC_CPU_FREQ_80M);
  Serial.begin(9600);
  Serial.println("HX711 Demo");

  Serial.println("Before setting up the scale:");
  Serial.print("read: \t\t");
  Serial.println(scale.read());			// print a raw reading from the ADC

  Serial.print("read average: \t\t");
  Serial.println(scale.read_average(20));  	// print the average of 20 readings from the ADC

  Serial.print("get value: \t\t");
  Serial.println(scale.get_value(5));		// print the average of 5 readings from the ADC minus the tare weight (not set yet)

  Serial.print("get units: \t\t");
  Serial.println(scale.get_units(5), 1);	// print the average of 5 readings from the ADC minus tare weight (not set) divided
						// by the SCALE parameter (not set yet)

  scale.set_scale(70.f);                      // this value is obtained by calibrating the scale with known weights; see the README for details
  scale.tare();				        // reset the scale to 0

  Serial.println("After setting up the scale:");

  Serial.print("read: \t\t");
  Serial.println(scale.read());                 // print a raw reading from the ADC

  Serial.print("read average: \t\t");
  Serial.println(scale.read_average(20));       // print the average of 20 readings from the ADC

  Serial.print("get value: \t\t");
  Serial.println(scale.get_value(5));		// print the average of 5 readings from the ADC minus the tare weight, set with tare()

  Serial.print("get units: \t\t");
  Serial.println(scale.get_units(5), 1);        // print the average of 5 readings from the ADC minus tare weight, divided
						// by the SCALE parameter set with set_scale

  Serial.println("Readings:");
}

void loop() {
  Serial.print("one reading:\t");
  Serial.print(scale.get_units(), 1);
  Serial.print("\t| average:\t");
  Serial.println(scale.get_units(10), 1);

  scale.power_down();			        // put the ADC in sleep mode
  delay(5000);
  scale.power_up();
}

amotl, 2019-07-23:
Dear @mggil7,

from looking at the constructor `HX711 scale(17, 16);`, we can deduce you are using a deprecated version of this library which is sensible to timing problems especially on newer and faster MCUs. See also #123.

We have been able to confirm the revamped version of this library works on similar hardware you might be using, so it is worth to give it a shot. The current version is available from this repository as well as the Arduino Library Manager.

With kind regards,
Andreas.

mggil7, 2019-07-23:
Dear Andreas, thank you for your prompt reply.

I have installed Lemio´s fork but unfortunately, reading still not consistent. I will keep trying and if I reach a positive outcome, will let you know.

Once again, thank you for your prompt reply.

Best Regards,

Miguel Gil

amotl, 2019-07-23:
Dear Miguel,

[Lemio´s fork](https://github.com/lemio/HX711) shows
```
This branch is 2 commits ahead, 50 commits behind bogde:master. 
```
so it might be outdated. Hardware support for the current library of @bogde has been verified for:

- ATmega328: Arduino Uno
- ESP8266: WeMos D1 mini, Adafruit HUZZAH
- ESP32: ESP32 DEVKIT V1, Heltec WiFi Kit 32, Adafruit Feather HUZZAH32
- STM32 F1 (Cortex-M3): STM32F103C8T6 STM32 Blue Pill Board

-- https://github.com/bogde/HX711#hardware-support

Aside from the MCU with appropriate software driver support, other things might go wrong as well.

As a first measure, I would recommend using the current library from here.

With kind regards,
Andreas.

mggil7, 2019-07-23:
Dear Andreas,

Thank you for your support. I tried with the updated library but the same fluctuations remains.

Please let me know if you have any additional suggestion.

I am more than willing to address this issue.

Best regards,

Miguel Gil

amotl, 2019-07-24:
Dear Miguel,

while I would like to see the fluctuations would come from the hardware in any way, we can't promise the library is actually ready for reading on the ESP32 without and glitches. We haven't been able to confirm it personally but trust others who reported this back to us.

When looking at the code, you might want to try again without `rtc_clk_cpu_freq_set(RTC_CPU_FREQ_80M);`, maybe you want to share your updated code again by putting it on https://gist.github.com/.

With kind regards,
Andreas.

mggil7, 2019-07-24:
Dear Andreas,

I had an issue with hardware and atom instalation. I will run it again and get back to you shorthly .

Regards,

Miguel Gil

mggil7, 2019-07-28:
Dear Andreas,

I have replaced the HX711 board and now it is running smoothly .

I have replaced the HX711 board and it is running smoothly .

I do appreciate your help and prompt attention. Thank you.

Best regards,

Miguel Gil

amotl, 2019-07-28:
Dear Miguel,

thanks for letting us know and happy weighing.

With kind regards,
Andreas.