bogde/HX711 #152: Non-blocking mode not working.

Curious-Nikhil, 2019-07-29:
The HX711 - timeout example sketch doesn't work.

It's supposed to return "0" when the loadcell is disconnected.  I still get garbage values.

[image]

Please help.

Curious-Nikhil, 2019-07-29:
added a temp fix.

amotl, 2019-07-29:
Dear @Curious-Nikhil,

thanks for writing in. Sorry to hear this mode added by us the other day doesn't work as expected for you. We have to admit that we haven't had the chance to test it on actual hardware yet.

I believe the fix would count as a workaround for you but we should really fix the library appropriately if possible.

With kind regards,
Andreas.

Curious-Nikhil, 2019-08-03:
Hey!
Thanks a lot for the amazing library. @amotl !

It's cool. No problem. 
Thanks again! :D

amotl, 2019-08-03:
I'd like to extend the thanks to @bogde and all others who helped the library getting where it is now. Glad this piece of software is helpful for you.