bogde/HX711 #159: connecting HX711 with ES01

isatsara, 2019-10-13:
Dear All. I am new to Arduino. So please help me to sort out my problem. I am not good in Electronics as well. 

Could you please tell me, how to use  ES01 (ESP8266 family) with HX711.
I tried to use
     pin **DOUT** **3**  // Pin connected to HX711 data output pin
     pin #define **CLK**  **5**  // Pin connected to HX711 clk pin
to connect with HX711 module. 
When i use the  **scale.begin(DOUT, CLK);** in my code, web server is not responding. if i removed all the codes which relevant to HX711, server is online back.

I know my code is messy, due to i copied codes from several places while testings :(. 
here i attached the code i used
[HX711-ES01.txt](https://github.com/bogde/HX711/files/3721563/HX711-ES01.txt)

Please help me on this. Thank you very much

bogde, 2019-10-21:
Does the example code that comes with the library work?

closed 2019-11-15 by bogde

isatsara, 2019-11-15:
Hi guys, Sorry i was unable to get in touch with the issue i raised. Finally i found the issue.
Issue with the connection between HX711 and the load cell. Reason was, i connected the load cell and the HX with a jumper cables in my test board. Then i realized the signal which received to the HX711 is weaker  when coming through the Jumper cables. Once i directly connect to the HX711 with the load cell it works.