bogde/HX711 #161: I'm testing HX711 with Arduino 33 BLE and it does not seem to work

morettigiorgio, 2019-10-21:
I'm testing an hardware with HX711 with holder for Arduino Nano/Micro series and i'm trying to switch from normal Arduino Micro and Arduino Nano 33 BLE Sense.
While with the Micro, the HX711 library responded with good and stable values, with BLE i see strange and unstable values.
In both cases, i used the power supply for HX711 at 3,3V
Do you have any suggestion to use this library with Nordic nRF52840 inside on Arduino 33 BLE?
Thanks

morettigiorgio, 2019-11-06:
The problem is in the clock out! With my oscilloscope I saw 25 pulse (correct for HX711 protocol) with Arduino Micro and about 100 pulse (incorrect) with Arduino 33 BLE. It seems that after a pulse clock, there are another 3-4 wrong pulse.
I tried to change the clock PIN with another (D2 D2 D6 A2 A3) but without success.
Do you have any suggestion?
Thanks

morettigiorgio, 2019-11-06:
I solved it,
i had an old resistor with a pull-up function on clock line and while the pin was low, unexpectedly this increased to rail. This was perceived as another clock pulse. After removing the resistance, everything went well.
Of course, i have to understand because this pin, in output mode (the library forces this state), has this behavior, like a float pin while it is low, but this is off topic

closed 2019-11-06 by morettigiorgio

itsybitsynoob, 2020-04-23:
I'm also trying to run the hx711 on a nrf52840 (adafruit itsybitsy m0 express) and I have not been able to get this library to work via arduino. I definitely suspect its something with the clk line. I'm very new to all of this, could you elaborate on what you had to change with the code to get the HX711 to read from your microcontroller?

morettigiorgio, 2020-04-24:
Hi @itsybitsynoob,
I am using [bogde/HX711](https://github.com/bogde/HX711) library and I must say that I have not modified this library. My problem was on hardware. I had used the external pullup resistor on the clk line which it had caused an abnormal behavior.
However, in my experience, I suggest you to do many attention to the interrupt inside [HX711.cpp](https://github.com/bogde/HX711/blob/master/src/HX711.cpp) (line [137](https://github.com/bogde/HX711/blob/master/src/HX711.cpp#L137) and [166](https://github.com/bogde/HX711/blob/master/src/HX711.cpp#L166)) because it is important for a good HX reading but it could cause other problems (in my case, it had blocked the ArduinoBLE library and I had to comment on it)

itsybitsynoob, 2020-04-24:
@morettigiorgio thank you. I saw the delay update in the “ spring cleaning” area of this library. Is the issue that the nrf52840 runs too fast and the interrupt is not long enough to tell the HX711 to power up/down or read data? 
I have zero programming knowledge and this is my first project, I had a little bit of luck using it with circuit python but there were very few resources for that so I decided to go with Arduino. 
I’m using the Adafruit itsybitsy express Bluetooth, I noticed some of the pins are slower, <10mhz would it be better to use these or to avoid them? Also I’m not using the onboard SDK pin and instead calling my own clock pin. 
If any of these things jumps out or rings a bell to you please do tell and I can zero in on those parts as well.

I appreciate your prompt response and you coming back to this old thread!

morettigiorgio, 2020-04-27:
Certainly my use is before on March of this year, but I don't see significantly different for you with the last commit. I commented out interrupt only because it gived problem in another library, no itself.
I find very strange that it doesn't work on ItsyBitsy M0 and ItsyBitsy 32u4, I would double check the hardware.
It is not important that you use a specific pins. The bus isn't I2C/SPI. All pins is right.
If you have an oscilloscope, you could look at the clock data difference between two circuits