bogde/HX711 #162: HX711 doesn't change value when weight it added; random spikes

itisyeetimetoday, 2019-10-27:
Hi,
Here is my project: https://drive.google.com/open?id=1Z3F06SedyUEUU3kLxyKrCXW0QrrVnwQ2
Here is my code
`#include "HX711.h"

HX711 cell(A0, A1);

void setup() {
  Serial.begin(9600);
}

long val = 0;
float count = 0;

void loop() {
  count = count + 1;
  
  // Use only one of these
  //val = ((count-1)/count) * val    +  (1/count) * cell.read(); // take long term average
  //val = 0.5 * val    +   0.5 * cell.read(); // take recent average
  val = cell.read_average(5); // most recent reading
  
  Serial.println( val /10000);
}`
The script returns 838 constantly with sudden spikes +- 400. Also, adding or removing weight doesn't change any of the values. I have tried using both digital and analog pins, changing the order of the pins in the script, and have tried using 2 different load cells. Should someone please help me resolve this issue? Thank you.

morettigiorgio, 2019-11-06:
excuse the banality, but the circuit pins do not seem to be soldered but only inserted on the strip. Are you sure you have a stable connection?
HX711 needs a stable clock for respond in correct mode...

bogde, 2019-11-15:
Did you try a different HX711? Maybe your board is bad.

isatsara, 2019-11-15:
@itisyeetimetoday 
i think  @morettigiorgio 's idea is more relevant to your issue.   according to your image, load cell and the HX711 is connected through some jumpers. 
 So try to connect Load cell and the HX711 directly. 
I experienced the same.
Note: No need to solder. but connect the load cell wires to the test board directly. It works for me.

Fusseldieb, 2021-05-21:
I have searched around for quite a bit and the problem seems to lie in the bitbanging. It isn't precise. `SPI.transfer` should be instead.

aspdigital, 2021-05-21:
> I have searched around for quite a bit and the problem seems to lie in the bitbanging. It isn't precise. `SPI.transfer` should be used instead.

I don't think "precision" of the clock matters, as the conversion is clocked by an internal oscillator and you use the digital interface only after the conversion is complete.

The guilty party may indeed be that internal oscillator, and an interesting test might be to use an external crystal oscillator. The data sheet is sketchy on how to connect one, though, and anyway the SparkFun board has XI grounded so you'll have to lift the pin.

Fusseldieb, 2021-09-28:
Issue still persists and I am running out of ideas...