bogde/HX711 #163: Arduino UNO + straight bar Load Cell (40 KG) + HX710B IC based Weighing scale

Vamshi253, 2019-11-12:
Hello,

I have been using these library files and sketches, but the readings are not accurate.  Even though there is no weight on load cell it's indicating non zero value but it's somewhat close to zero and whenever I place the 200 gram weight on load cell, its indicating 195.67 or 196.67 or something close to 200 but not accurate enough. whenever I slide the weight and put it at a different position on the load cell, the readings are changing.

It's a humble request to all of you to help me in this regard considering to get accurate readings.

Thank you.
Vamshi

gsaplx, 2019-11-14:
Hi all,

I'm also struggling with the weight increasing. If you find a fix please let me know

bogde, 2019-11-15:
I'd guess it's more of a mechanical problem, maybe "load cell creep" https://www.zemiceurope.com/en/blogs/creep.html