bogde/HX711 #164: HX711 and two arduinos

marcinolokk, 2019-11-21:
Hi and best regards to the creators!

This is my first post on Github so please forgive me my mistakes :)

I am trying to make one Arduino work alongside the other one. I need both of them to get the readings from HX711. Apparently, when I split up the output of the hx711 to both Arduinos, they have interfered readings (randomly fluctuating). The data is useless. Here is my question, is there a way I can connect one output signal of hx711 to two devices (In my case Arduino)?

Best Regards

bogde, 2019-12-02:
I doubt you can do that, at least without using additional components and synchronizing readings from the two Arduinos. Why don't you just read the HX711 using one Arduino and then share the value with the other one via some kind of connection like serial, SPI etc.?

closed 2021-03-04 by bogde

marcinolokk, 2021-03-04:
Sorry, somehow I missed your answer. That's how I actually solved the issue!