bogde/HX711 #167: Find the correct calibration factor at powerup using a know weight ?

hjojohan, 2019-12-27:
Hi 
I would a code example that could find out the calibration factor at Arduino powerup using a known weight type 1 kg. The initializing code should loop by itself until the calibration factor is found and the store the factor in eeprom on a arduino.   From my experience the SCALE needs recalibration everytime you move it, I send them by mail to a another site. 

For example : Use a LED to signal (flashing) to inform enduser that the Arduino is in 'calibration mode' and a known weight should be placed on the scale and when calibration is done by code the LED light should turn solid and after this the scale is ready for normal operation

Is there a kind soul out the that could give me sample code to look at ?

simpat1zq, 2020-02-02:
Not sure if it would help you. But I made a web interface for the ESP8266 that handles calibration with a wizard.:

https://automatedhome.party/2019/12/14/the-filaweigher-a-standalone-wifi-web-based-weight-sensor-for-3d-printer-filament-for-less-than-7/

bogde, 2021-03-04:
@simpat1zq Thanks for your help!

Closing due to lack of additional feedback.

closed 2021-03-04 by bogde