bogde/HX711 #172: hx711 module design fault fix

konosubakonoakua, 2020-02-28:
Solution: Use A wire to connect **E-** and **GND**
[image]
I read the datasheet of **HX711**, and found a design fault of the hx711 moudule in the market.
[image]
In the picture above, we can see that the **pin-3** is **AGND**, 
The picture below shows the recommended application in it's datasheet.
[image]
It should be connected to **GND**, but the module dosen't do that.
[image]

amotl, 2020-02-28:
Dear @konosubakonoakua,

many breakouts out there are of poor quality, see also [1] and [2]. It's sad but true.

Cheers,
Andreas.

[1] https://community.hiveeyes.org/t/analyse-und-diskussion-um-eine-korrekte-integration-des-hx711/2802
[2] https://community.hiveeyes.org/t/stromversorgung-hx711/893

amotl, 2021-03-04:
Hi Bogdan,

while this topic is not about an issue of the library, but about the hardware, the information here is still very valuable for the community. In that manner, I suggest to add a tag called `information` to this issue and keep it open for the sake of discoverability. I will delete this comment afterwards.

With kind regards,
Andreas.

bogde, 2021-03-04:
@amotl I was just looking at this today and while i can confirm the reported issue (E- AGND not connected to GND), actually connecting AGND to GND didn't make a significant difference in my case. Nevertheless, it's something to keep in mind and something that can potentially help others. I added a label to this issue, I'm not sure if that's what you meant by tag, let me know.

I was also thinking to add this to the FAQ file I just created, especially if @konosubakonoakua agrees to let me use his post and images.

I was also looking at #148 
I know it's been some time since that was posted but i'd appreciate your thoughts on it. I've been doing tests lately and, as expected, that does make readings more stable. I was thinking to add some new methods that would allow the user to basically configure the library to set a number of bits to 0 in the read() function. Something like:
```
data[0] = data[0] && 128
```

konosubakonoakua, 2021-03-04:
@bogde sure, you can use them freely

bogde, 2022-06-02:
yes, that's what i meant but the change was never implemented

amotl, 2022-06-02:
Hi Bogdan,

> I was also thinking to add this to the FAQ file I just created, especially if @konosubakonoakua agrees to let me use his post and images.

I think you can also use the images at [1] shared by @msweef if you like.

With kind regards,
Andreas.

[1] https://community.hiveeyes.org/t/stromversorgung-hx711/893/17