bogde/HX711 #173: exit status 1 no matching function for call to 'HX711::HX711(int, int)'

sluke17, 2020-03-17:
So i've downloaded the header file for the strain gauge and have followed this tutorial by Electronic Clinic:

https://www.youtube.com/watch?v=cztWFrd-gQA

But I get the following error:
exit status 1 no matching function for call to 'HX711::HX711(int, int)'

I have tried declaring them as integers but the Arduino IDE moans at me when trying to verify.

Any suggestions?

amotl, 2020-03-21:
Dear @sluke17,

please have a look at the example [1]. The new interface is that the `begin(...)` method will obtain the pin parameters.

With kind regards,
Andreas.

[1] https://github.com/bogde/HX711/blob/master/examples/HX711_basic_example/HX711_basic_example.ino#L7-L12

amotl, 2020-03-22:
@AyamTepihit observed the same problem and reported about it within #175.

Rajanayaki, 2020-06-24:
Faced same problem.
I just replaced  HX711 scale(DOUT,SCK); with HX711 scale;
It worked fine. 

Hope this helps :)

navin-153, 2021-02-26:
I just replaced HX711  scale(int,int); with HX711 scale.just cleared the (int,int)

bogde, 2021-03-04:
Just for clarity, the library was changed at some point and some examples and tutorials still use the old way of initializing the pins. I created a FAQ file and listed this issue first, together with the solution [1], which has been correctly explained above as well.

[1] https://github.com/bogde/HX711/blob/master/doc/faq.md

closed 2021-03-04 by bogde

dsmithwc04, 2021-09-13:
I'm struggling to get my to work correctly.  I was getting the same error as above and applied the recommended fix and stopped getting any errors.  My problem now is once it displays "Servo Torque Measurment" in the serial monitor I don't get any text telling me to press "s" or "h".  Afrer a few minutes I went ahead and pressed "s" and hit send button and nothing happened.  Then a few moments later I get a slight movement of the servo and then it says "Initialization Complete" and then nothing.  Any ideas would be greatly appreciated!  Thanks.