bogde/HX711 #175: no matching function for call to 'HX711::HX711()'

AyamTepihit, 2020-03-22:
Issue persist even after following the example given.

AyamTepihit, 2020-03-22:
#include <HX711.h>

const int LOADCELL_DOUT_PIN = 3;
const int LOADCELL_SCK_PIN = 2;

HX711 scale;

example of the code i used. the error occur on the line HX711 scale;

AyamTepihit, 2020-03-22:
[image]

image included is the code i used exactly in the example

closed 2020-03-22 by AyamTepihit

amotl, 2020-03-22:
Dear @AyamTepihit,

will `#include <HX711.h>` work for you? I see you closed the issue alread: Did things start working for you?

With kind regards,
Andreas.

AyamTepihit, 2020-03-22:
Currently I am using other library works for my coding. I already tried changing into #include <HX711.h> still no luck, for now other library works fine, maybe I’ll fix my issue with your library soon after I calibrate my load cell with other library first.

Thanks anyway for helping my issue. I’ll contact you if there is another issue.