bogde/HX711 #179: Scale without tare

DVB-Arduino, 2020-04-01:
Hey. I make scales on the ESP8266. The system will go to sleep and then reboot. Therefore, I need to ensure that there is no taring at startup, because the weight at each reboot will be 0. I want to calibrate the system and at each start I need to display the weight that will be added from the moment of calibration. How to do it?
The problem is that when you turn on the system, the balance already shows some value. Therefore, calibration is used first calibration. 
Is it possible to specify a value in the taring function, which the system will initially take away, but not reset to 0?

bogde, 2020-04-01:
just don't call scale.tare() in your setup code.

MathisDu, 2020-05-02:
Hello, I have the same problem as DVB-Arduino, I'm doing a connected scale for a beehive with an ESP8266. However, when it reboots, the weight is not the same as before, whereas there is no scale.tare() in my setup code.

bogde, 2020-05-03:
if you properly calibrate once and then store the calibration value in code, and dont call tare() you should get proper weight after reboot. am i missing something?

jasonacox, 2020-06-14:
The key is to determine the calibration values (divider and offset) and set them in your code.  Do not use tare().  You will have something like this:

```c
  scale.begin(LOADCELL_DOUT_PIN, LOADCELL_SCK_PIN);
  scale.set_scale(98.11); // divider
  scale.set_offset(-125538);  // offset
```
Trying to determine the right values to use isn't easy.  I determined the *offset* by running the `read_average()`:
```c
  Serial.print("AVERAGE = ");
  Serial.println(scale.read_average(100));

  >>>>>>   AVERAGE = -125538
```
The *divider* is a bit more difficult but is covered in the README under section "How to calibrate your load cell".  Basically, find a known weight (e.g. 1kg weight) or something you can weight on a trusted scale (and record the true weight as w1).  Place the weight on the loadcell and call `get_units()` to get the measured weight (w2).  The *divider* value will be = w2/w1.

```c
void loop() {
  if (scale.is_ready()) {
    long reading = scale.read();
    Serial.print("HX711 reading: ");
    Serial.print(reading);
    Serial.print(" - and get_units : ");
    Serial.println(scale.get_units(10), 2);
  } else {
    Serial.println("HX711 not found.");
  }
  delay(100);
}
```
Add the set_scale(*divider*) and set.offset(*offset*) to your code and you should get the same results even after a power cycle.

s56jsc, 2021-12-05:
I use simple solution and work great
First I read zero factor:
`long zero_factor = scale.read_average(); `

This is my offset. Then I put this value in:
`scale.set_offset(zero_factor);`

jakobrim121, 2022-06-23:
Hi,

I am using a fixed resistance (350 Ohm each) Wheatstone bridge, the load cell and Arduino Mega 2560. The Arduino gives a random ADC value of -963250. I am using the set calibration function on that value and tare after that. Now when I am unbalancing one of the leg of Wheatstone bridge by 5 Ohm (350 -> 345) I am getting a very high ADC Value.
HX711 reading: 6977316
HX711 reading: 6977258
HX711 reading: 6977327
HX711 reading: 6977260
HX711 reading: 6977281
HX711 reading: 6977297
HX711 reading: 6977305
HX711 reading: 6977340
HX711 reading: 6977299
HX711 reading: 6977383
HX711 reading: 6977331
HX711 reading: 6977317
HX711 reading: 6977332
HX711 reading: 6977387
I don't know why? Can anyone please help.