bogde/HX711 #180: Adapt libary to HX710

elfelfelfelf, 2020-05-21:
This library easily works with the HX710 for basic functionality. However because on the HX710 the setup and config of the chip is not done in hardware but in software, the library would have to be updated. From what I understand all of the config is done by sending different amounts of clock pulses when reading. These are the different configs:

PD_SCK Pulses | Input | Data Rate
-- | -- | --
25 | Differential | 10Hz
26 | DVDD-AVDD | 40Hz
27 | Differential | 40Hz

Maybe someone could add that functionality...

bogde, 2020-05-21:
i think you can already use the set_gain functionality for that;

set_gain(128) should give you 25 pulses,
set_gain(32) should give you 26 pulses, and
set_gain(64) should give you 27 pulses

not the cleanest way, but should work. unfortunately i dont have a 710 board to test

elfelfelfelf, 2020-05-24:
Yes! The set gain function worked as expected... Thanks a lot

closed 2020-05-24 by elfelfelfelf

AlbiBone, 2020-08-05:
Hi @elfelfelfelf ,
I am trying to use HX710b so I am interested in this post. 
I did not understood why you written that with 26 SCK pulses, you get DVDD-AVDD	40Hz.
If I understood correctly from the [datasheet](http://read.pudn.com/downloads660/doc/comm/2680906/hx710b_english.pdf), with 26 pulses, you get the temperature of the chip. Correct?

elfelfelfelf, 2020-08-05:
@AlbiBone what you say is correct, but there are two versions of the HX710. One ends on 'a', the other on 'b'. They are basically the same chip with the one difference that the 'a' revision has a built-in thermocouple, while 'b' has DVDD-AVDD.
If you send 26 pulses you get:
-either the reading of the thermocouple (if you have the 'a' version) at a sample rate of 40 hz
-or the DVDD-AVDD (if you have the 'b' version) at a sample rate of 40 hz
Hope that makes things more clear
See this datasheet for further reference:
[https://img.ozdisan.com/ETicaret_Dosya/470782_5454613.PDF](url)