bogde/HX711 #181: Increasing readings with HX710b

10git, 2020-05-23:
I'm using the [HX710b pressure sensor](https://content.instructables.com/FF0/EBBK/K9UE5QM5/FF0EBBKK9UE5QM5.LARGE.jpg) hooked on Arduino Uno R3. Arduino is powered by the USB cable.

When running the [HX711_full_example](https://github.com/bogde/HX711/blob/master/examples/HX711_full_example/HX711_full_example.ino) I get increasing values at each reading:

> HX711 Demo
> Initializing the scale
> Before setting up the scale:
> read: 		4216884
> read average: 		4217088
> get value: 		4217178.00
> get units: 		4217341.0
> After setting up the scale:
> read: 		4217602
> read average: 		4218342
> get value: 		1828.00
> get units: 		1.0
> Readings:
> one reading:	1.1		| average:	1.4
> one reading:	5.1		| average:	5.5
> one reading:	9.6		| average:	10.0
> one reading:	13.6	| average:	13.8
> one reading:	15.9	| average:	16.1
> one reading:	18.6	| average:	18.9
> one reading:	22.9	| average:	23.2
> one reading:	26.7	| average:	27.0
> one reading:	29.5	| average:	29.7
> one reading:	32.6	| average:	32.8
> one reading:	35.8	| average:	36.1
> one reading:	39.2	| average:	39.4
> one reading:	41.9	| average:	42.0
> one reading:	43.0	| average:	43.1
> one reading:	44.0	| average:	44.1

I set different gain values, as you suggested in the issue https://github.com/bogde/HX711/issues/180, but I still have the same bad behaviour.

**Strange thing**: the device gives me readings even when I disconnect its GND from the Arduino's GND:

> HX711 Demo
> Initializing the scale
> Before setting up the scale:
> read: 		4324590
> read average: 		4324229
> get value: 		4323756.00
> get units: 		4323666.0
> After setting up the scale:
> read: 		4323619
> read average: 		4323624
> get value: 		108.00
> get units: 		0.1
> Readings:
> one reading:	0.1		| average:	0.1
> one reading:	1.1		| average:	1.2
> one reading:	1.6		| average:	1.7
> one reading:	2.6		| average:	2.8
> one reading:	4.1		| average:	4.3
> one reading:	5.9		| average:	6.1
> one reading:	7.5		| average:	7.6
> one reading:	97.3	| average:	84.7 <= GND was disconnected here
> one reading:	106.8	| average:	93.4
> one reading:	111.8	| average:	98.1
> one reading:	115.5	| average:	101.9
> one reading:	118.6	| average:	104.8
> one reading:	33.4	| average:	32.2 <= GND was reconnected here
> one reading:	27.6	| average:	27.5
> one reading:	26.3	| average:	26.2

Some more measures show that:

1. The whole circuitry draws arround 12mA, even when _power_down()_ is called. I also tested with an Arduino Mini Pro 8MHz 3.3V and it draws 870µA all the time (serial was not used in this case).
2. The whole circuitry draws arround 11mA when _power_down()_ is called **and** when the GND is disconnected as shown above. I also tested with an Arduino Mini Pro 8MHz 3.3V and it draws 870µA during readings, then 5.6µA after _power_down()_ was called. Again, the device shall run with the GND disconnected in order to see the drop from 870µA to 5.6µA (serial was not used in this case).

elfelfelfelf, 2020-05-24:
I have had the same issue and worked out how to do it: You have to check if the sensor is ready to spit out the data and as soon as it is, you can read the data. Of course your sampling rate is limited by how fast the Arduino runs through one "loop", but using the minimal code below I got very close to 40 samples a second. If you wanted to take it further you could figure out how to have the Arduino trigger an interrupt on the falling clock signal of the HX710 and then read the data... Hope that helps

```c++
#include "HX711.h"

const int LOADCELL_DOUT_PIN = 2;
const int LOADCELL_SCK_PIN = 3;

HX711 scale;

void setup() {
  Serial.begin(57600);
  scale.begin(LOADCELL_DOUT_PIN, LOADCELL_SCK_PIN);
  scale.set_gain(64);
}

void loop() {
  if (scale.is_ready()) {
    Serial.println(scale.read());
  }
}
```

10git, 2020-05-25:
The check you mention is already done in the _read()_ method. Unfortunately, as expected, when running the minimal code, the values are still increasing over the time. Even for the interrupt version of your code I still have the same odd behaviour.

```
#include "HX711.h"

const int LOADCELL_DOUT_PIN = 2;
const int LOADCELL_SCK_PIN = 3;

HX711 scale;

void setup() {
  Serial.begin(57600);
  scale.begin(LOADCELL_DOUT_PIN, LOADCELL_SCK_PIN);
  scale.set_gain(64);
  attachInterrupt(digitalPinToInterrupt(LOADCELL_DOUT_PIN), get_it, LOW);
}

void loop() { }

void get_it() {
  Serial.println(scale.read());
}
```

tamilCoder, 2020-07-13:
Im too having a similar kind of an issue, Please share if you find the solution and ill share if i find one.

nameiskittu, 2023-02-01:
i want to convert this values into pressure units, can anyone explain how to do it

ba05, 2023-02-01:
> i want to convert this values into pressure units, can anyone explain how to do it

Zero out, apply full scale pressure, note output. Output divided by full scale pressure is calibration coefficient.