bogde/HX711 #183: Nodemcu with HX711 gives constant value

Rajanayaki, 2020-06-24:
Hello, I am a newbie to arduino and nodemcu things.I am in need of help where i am using a ESP8266 12E nodemcu with 1kg load cell sensor to get the weight value . Actually,it worked fine with arduino uno. Yet, i cant pass that weight value to nodemcu. So i tried connecting Nodemcu directly with loadcell. It gives constant weight. No change even when weight applied. FYI, am using arduino 1.8.12 IDE

Code:

#include "HX711.h"
#define DOUT  D2
#define CLK  D3

HX711 scale;

float weight; 
float calibration_factor = 954310 ;//This worked for my arduino Uno
void setup() {
  Serial.begin(9600);
  Serial.println("HX711 scale demo");
  scale.begin(DOUT, CLK);
  scale.set_scale(calibration_factor); 
  scale.tare(); //Assuming there is no weight on the scale at start up, reset the scale to 0

  Serial.println("Readings:");
}

void loop() {
  Serial.print("Reading: ");
  Serial.print(scale.get_units(), 1);
  scale.set_scale(calibration_factor); //Adjust to this calibration factor
 
  weight = scale.get_units(5); //5
   Serial.print("kg"); //You can change this to kg but you'll need to refactor the calibration_factor
  Serial.println();
}

[image]

I changed the baud rates to 115200 .Got no positive results.
Also, since this worked fine with arduino Uno,I tried sending the arduino output to nodemcu using serial communication. But the vice versa worked but not arduino to nodemcu communication. 
Does that mean my nodemcu isnt getting any input values? Any faults?

bogde, 2020-06-25:
Are you sure your Nodemcu wiring is correct? Can you post a picture or schematic of your wiring?

Rajanayaki, 2020-06-28:
[image]

**NODE MCU 1.0**    -    **HX711**
VIN(6-12V)           -   VCC
GND                     -   GND
D2                       - DOUT
D3                       -     CLK

karansoi, 2020-07-02:
@Rajanayaki, I observe that you are powering the HX711 board from the node mcu. I presume that this is 3.3 V. If you observe the sparkfun HX711 [https://www.sparkfun.com/products/13879](url)
you will notice it has pins VCC and VDD. The VCC is for 5V and VDD for 3.3 V this is for the digital part of the HX711 chip.
You will have to isolate this DVDD pin from the track on the board and connect it to 3.3 V frm the node mcu. VCC and VDD have to be separated. Also please refer to the schematic too
[https://cdn.sparkfun.com/assets/f/5/5/b/c/SparkFun_HX711_Load_Cell.pdf](url)
Hope this helps.

karansoi, 2020-07-04:
> [image]
> 
> **NODE MCU 1.0** - **HX711**
> VIN(6-12V) - VCC
> GND - GND
> D2 - DOUT
> D3 - CLK

Please check if the -E pin where you connect the load cell is grounded, some of these green HX711 boards come without this ground connection.

bangpradyumna, 2020-09-22:
> @Rajanayaki, I observe that you are powering the HX711 board from the Nodemcu. I presume that this is 3.3 V. If you observe the sparkfun HX711 [https://www.sparkfun.com/products/13879](url)
> you will notice it has pins VCC and VDD. The VCC is for 5V and VDD for 3.3 V this is for the digital part of the HX711 chip.
> You will have to isolate this DVDD pin from the track on the board and connect it to 3.3 V from the Nodemcu. VCC and VDD have to be separated. Also please refer to the schematic too
> [https://cdn.sparkfun.com/assets/f/5/5/b/c/SparkFun_HX711_Load_Cell.pdf](url)
> Hope this helps.

Hey,
There are a few tutorials like this : https://thenewstack.io/off-the-shelf-hacker-feel-force-with-a-load-sensor    which have the exact same connections as OP i.e HX711 is powered directly by Vin of nodemcu. How does it work in their case ?

I also did the same and my code just gives 0 as a result no matter what load is put on the load cell. I'd appreciate any help.

Thanks

bierzorutas, 2021-04-28:
Same problem like bangpradyumna is mentioning here, tried also that example of thenewstack.io and it doesn´t work, always gives me 0

aspdigital, 2021-04-28:
While this is an old issue, it's worth noting that the HX711 data sheet specifies the range for VSUP and DVDD as 2.7 V to 5.5 V. The Sparkfun board connects its VCC pin to VSUP and to the emitter of the pass transistor. So there is no reason why you can't run the HX711 on a single 3.3 V supply from your micro board.

HOWEVER -- RTFDS. 

> When using internal analog supply regulator, the dropout voltage of the regulator depends on the external transistor used. The output voltage is equal to VAVDD=VBG*(R1+R2)/ R1 (Fig. 1). This voltage should be designed with a minimum of 100mV below VSUP voltage.

(There is a discrepancy between the HX711 data sheet and the SparkFun breakout board schematic. The former uses the equation given above. The latter has R2 in the denominator. At least one person commented on the SparkFun product page, and it appears that the SparkFun schematic is correct.)

100 mV is the HX711's internal regulator's drop-out spec. That is, the regulator input must be at least 100 mV higher than the output, or the regulator doesn't work. (What "doesn't work" means is not defined.) 

The Sparkfun board, and I assume every design that just follows the data sheet standard, uses the internal regulator with the external pass transistor. Indeed the Sparkfun data sheet has the equation for AVDD right on it. HX711 data sheet says that VBG = 1.25 V. With Sparkfun's values for R1 and R2, and VSUP = 5 V, AVDD is 4.3 V, and that difference of 700 mV easily meets the 100 mV drop-out requirement.

Set VCC (that is, VSUP) to 3.3 V. The calculation for AVDD remains the same, and obviously AVDD = 4.3 V simply won't work with VSUP at 3.3 V. The feedback resistor needs to be re-scaled to work at the lower input voltage.

So pick something: AVDD = 3.0 V. This gives R1 = 11.5k and R2 = 8.2k, so change R1 from 20k to 11.5k and the regulator should work and you should get non-zero conversion results.

Rajanayaki, 2021-04-28:
I would like to add what I did to solve the issue. The simplest I could think off to solve this was to make Arduino as a Master (since my HX711 worked well with Arduino) and NodeMCU as a slave . 

Here is the tutorial I used : 
https://www.hackster.io/RoboticaDIY/send-data-from-arduino-to-nodemcu-and-nodemcu-to-arduino-17d47a

Hope this helps.

adrian219, 2022-04-16:
I had the same problem. I used Arduino Uno and it worked (i pinned to 5V power in board) but the same connection to ESP8266 (Nodemcu V3) shown constant value.

I changed the power pin in HX711 from VCC to VDD and started working.

I hope these information can be helpful for others. :)