bogde/HX711 #187: Correct reading on Setup but not on Loop

meiralon, 2020-08-20:
Hello 

I am running a code on multiple sensors.
and at the setup in the Arduino I run :

scale.set_scale();
scale.tare();  //Reset the scale to 0
scale.set_scale(calibration_factor);
Serial.print(scale.get_units(), 1);

and at the loop in the arduino I run only :
Serial.print(scale.get_units(), 1);

but I get different result ,like there is some kind of bias.
the scale is measuring but with offset.

thank you
Meir

scorbesero, 2020-08-20:
I have had success using the following in my setup:

`scale.begin(DOUT, CLK);
  scale.set_scale(calibration_factor); //This value was obtained by using the SparkFun_HX711_Calibration sketch
  scale.tare(); //add stabilizing time to tare 

float measured_weight = 0.0; //measured weight
`

For the loop, try something like this:

`measured_weight = abs(scale.get_units(5)); //takes the average of five readings, adjust to your preference
Serial.print(measured_weight, 1);`

bogde, 2021-03-02:
Did you find a solution? Otherwise, can you post the rest of your code and give more details about your setup?

meiralon, 2021-03-03:
Hello Bogde
Thank you for your message.
i didn't solve this issue, but i added a shift in the loop in order 
to overcome it. see explanation as followed.
i will be happy for your expanation regarding this issue.

The code at the Arduino Setup is as followed:

scale.set_scale();
scale.tare();  //Reset the scale to 0
scale.set_scale(calibration_factor);
Serial.println();

Serial.print("Sensor ");
Serial.println(i+1);
Serial.print(":");
delay(5);
Serial.print(scale.get_units(), 1);
Serial.print(" kg"); 
Serial.println();

I read 0 Kg.
at the Arduino loop i get different reading, when i run:

Serial.print(scale.get_units(), 1);

in order to correct it i use set_offset :

scale.set_offset(zero_factor);

and this brings it to zero.

Thank you 
Meir

meiralon, 2021-04-06:
Hello bogde

Do you have an idea why it is happening ?

Thank you

Meir

bogde, 2021-04-06:
why dont you follow the example and use:
```
scale.set_scale(calibration_factor);
scale.tare();	
```

is there any reason why you're using set_scale without a parameter?

meiralon, 2021-04-07:
Hell Steven

This is what run, but it seems when it turns into the void loop, the offset information
is lost.
what do you think  ?

Meir

bogde, 2021-04-07:
i think you should stick to the example code until you figure it out

theres no reason to call set_offset manually, since i see you're already trying to use tare, which basically sets the offset for you

take a look at src/HX711.cpp to see how tare works

meiralon, 2021-04-14:
Hi Steven

Is it possible to change the clock speed (PD_SCK) or the high time of the pulse?
I Have seen at data sheet of the HX711 , it can be changed.
Thank you
Meir