bogde/HX711 #190: Serial Monitor open required to get scale.get_units(20)

stevensteichen, 2020-10-26:
WEMOS D1 Pro Mini
HX711
4 load cells in half bridge
Sends weight to ThingSpeak channel.
USB power from Laptop
Serial Monitor Open. Everything works great.
Close Serial Monitor window, all weights drop to -26.62845 (arbitrary as it's get_units) no matter what is on the scale. Open Serial Monitor and it resumes with correct weight. No restarts that I can see.

I do not have any While (!Serial), I have no serial at all on my side.
The sketch runs and send 'data' (good or bad) to ThingSpeak without any indications of a WDT as my delays between reads is 20 seconds so I should see a 40 second hole if it restarts and I don't. In the Arduino IDE, Open or Close monitor is all that it takes, it's a switch.

The only change is Serial Monitor open/works or close/fails. No reset required. Serial should not, normally is not, required so why is the sketch requiring Serial Monitor? Is it confusing serial to HX711 and Serial to Serial Monitor? They use 2 different clients and should be two different objects.

I considered a reset issue and needing to pull the reset pin but I see no indication resetting.

#include "HX711.h"
#include "ThingSpeak.h"
#include <ESP8266WiFi.h>

//thingspeak server
char server[] = "api.thingspeak.com";

HX711 scale;

  char ssid[] = "****************";    //  your network SSID (name) 
  char pass[] = "****************";   // your network password
  int status = WL_IDLE_STATUS;
  WiFiClient  client;
  
unsigned long myChannelNumber = 502530;
const char * myWriteAPIKey = "****************";

void setup() {
  // parameter "gain" is ommited; the default value 128 is used by the library
  // HX711.DOUT	- pin #A1 D3
  // HX711.PD_SCK	- pin #A0 D4
  //scale.begin(A1, A0);
  scale.begin(D3, D4);
  
  scale.set_scale(10440.f);
  double tare = 278000; //raw 'scale.read_average(20)'  with no weight. we cant tare else we lose everything if watchdog timer, reset or power loss
  scale.set_offset(tare); //set tare hardcoded to base tank
  //scale.tare();				        // reset the scale to 0, do not use this!!!!

  WiFi.begin(ssid, pass);
  ThingSpeak.begin(client);
}

void loop() {

  ThingSpeak.writeField(myChannelNumber, 1, scale.get_units(20), myWriteAPIKey);

//rem sleeps for testing, makes no difference
  //scale.power_down();			        // put the ADC in sleep mode
  delay(20000);                                       // Thingspeak limits updates. Once it's working, i'll test for lowest weight every 24 hours and send only that value.
  //scale.power_up();
}

stevensteichen, 2020-11-06:
After much troubleshooting, my issue is DT pin is not staying low long enough <2uS. If Serial Monitor is open, DT gets pulled low by HX711 , <80mv and stays there as expected. If I close Serial and single shot DT, it recovers to 2.08V within about 2uS. Now, I messed up, I used Wemos D3 for DT and D4 for SCK. D3 is esp8266 GPIO0 used for DTR. I think that's the connection. Something in the SIL2104 UART chip is strong enough to allow the HX711 to pull the pin low. Without that chip sinking current, the HX711 can't sink enough to keep the pin below the required LOW of 2V long enough to flap the sck and get the data.

Can i do something to the HX711 DT/D3 pin or should I bodge wire and change the code to use GPIO4/D2 for DT?

If anyone is reading this, thanks for the time.

bogde, 2021-03-02:
Thanks for reporting your findings.

closed 2021-03-02 by bogde