bogde/HX711 #191: HX711 scale.get_units(10) return fluctuating values

bprovvis, 2020-11-06:
hello
with arduino nano - HX711- load cell
without weight on the scale or with weight

sample without weight on the scale
scale.get_units(10) return fluctuating values
reading -2407.70
reading -2424.80
reading -2681.20
reading -3088.90
reading -3432.30
reading -3619.40
reading -3834.00
reading -3943.30
reading -3867.50
reading -3767.70
reading -3680.80
reading -3667.60
reading -3786.70
reading -3930.50
reading -4144.80
reading -4473.50

i don't know how to calibrate my scale
-------------------my code
#include "HX711.h"
HX711 scale;
void setup() {
  Serial.begin(9600);
scale.begin(A1, A0);
  scale.set_scale();
  scale.tare();               // reset the scale to 0
}

void loop() {
  Serial.print("reading ");
  Serial.println(scale.get_units(10), 2);
  delay(2000);
}
Arduino Nano                   -------> HX711
    5V-                    VCC                A+     WHITE
    A0-                    SCK                A-      GREEN
    A1-                     DT                 E-       BLACK
    GND -               GND               E+      RED

thanks for any answer

bogde, 2021-03-02:
Did you solve this? What board are you using? It has been reported that some boards (like the Sparkfun one) work better and are more stable than the cheaper ones on Aliexpress and the like.

nedim78, 2021-06-06:
I have been plagued by this issue from march 2020...I was using hx711 in 2019 in about 10 beehive-scales  without any problems. But last year I bought 10 "green" breakout boards and they all had horrible drift like this shown above. I tested everything, from hardware modifications, arduino uno 5V, mini pro 3.3V, every solution I could find online, without success. So, I thought I bought a faulty batch, and that's it. So I went online, bought another batch of "violet" breakout boards with totally different design - same problems. So, I'm on the verge of abandoning hx711 alltogether. I now have 2 totally different boards, Arduinos Uno and MIni pro, 3.3V and 5V, and I can try any bright ideas :)

MBENKI, 2021-06-11:
@nedim78 Yes i got the same problem, but the values differ very slowly over time, i have no idea if it is the Arduino board o the HX711 chip or the weight sensors ... if i leave an object on the scale after some time i get a different weight for example 1000g becomes 1050g after few hours or a day, and can go to 950g or even 1100g which completely bugs me :s