bogde/HX711 #192: HX711 interface with esp32

Jayendrarnaik, 2020-12-09:
hello ,,,
i tried interfacing 20kg load with esp32 uisng hx711 module, but am not able make it correctly. I programed esp32 with basic example but hx711 is not getting detected. Tried the same hx711 with arduino mega adk ...its working prefectly.
esp23 pin:
dout-25
sck-22
(tried with almost all the pins avilable in eps32 :( )
[image]
[image]
please help....

bogde, 2021-03-02:
Did you solve this? The library has been tested with ESP32 and it worked. Are you sure your pins are correctly set up? Can you test the pin definitions with something else, like an LED, just to make sure the wiring is right?

bogde, 2021-04-26:
Closing due to lack of further feedback.

closed 2021-04-26 by bogde