bogde/HX711 #196: What is HX711 Output?

EnriqueVerduzco, 2021-02-18:
Hello, I was curious and wanted to know if someone here could help. 

What exactly is the output of the **read()** function when there is no tare or offset. I believe this is the mV (millivolt) value from the load cells connected to the HX711 but am not 100%. 

I am asking because I am trying to calculate the mV/V (millivolt per volt) alongside the weight of the scale but am stuck on how exactly to calculate the mV/V.

mcmahad, 2021-07-30:
mV/V is something that needs two measurements to compute.  Typical HX711 applications only measure one of them.

mV  usually comes from reading the A channel.  You get the result in ADC counts, but you can determine the conversion to mV based on knowledge of the chip and gain values selected.  It doesn't really matter if you decide to leave the result in ADC counts, however.

The 'Volts' part in the denominator is the problem.  Most applications don't measure this and the typical HX711 eval boards from places like SparkFun don't allow you to easily get access to it.  You need to run a typical measurement cycle but measure the excitation voltage instead of the bridge voltage.

There is a second analog channel on the HX711, I may try to connect it to see if I can compute mV/V from direct measurements.  The voltage on it needs to be divided down by resistors to be in range of the ADC for channel B.  Channel B also has a different gain value than channel A, so be careful of that.  It is discussed in the data sheet.

If you have both ADC counts from reading the excitation voltage and the bridge voltage, then you can simply divide the two numbers to get mV/V.  You will have to watch all the details for things like rounding/truncation/overflow, etc.  I would suggest doing this math in floating point because of the ranges of numbers.  I don't typically like floating point for most things in embedded projects, but I think this is one place that needs it.

Simplistically speaking, you could simply use the formula:

mv/V = chanAreading_AdcCnts / chanBreading_AdcCnts * 1000 * dividerResistorRatio

Multiplying by 1000 converts ADC counts to milli-ADC counts and the divider resistor ratio is the correction for converting 5 volts of Excitation down to 0.5 (or whatever) to be in range for the channel B measurement on HX711.  You should individually measure the divider resistors before you attach them to the board so you can know that value if it is really important to you.

ba05, 2021-07-31:
mV/V calculation: [Link](https://github.com/bogde/HX711/issues/208#issuecomment-890373362)