bogde/HX711 #199: How do I find the Slave Address for HX711

YudiNz, 2021-04-26:
Has stated in title, I am trying to make this sensor work with myRio from LabVIEW but am unable to find the slave address in order to access the device. 

Can anyone help me with this issue?

bogde, 2021-04-26:
i don't think i understand your question. however, here's a link to the datasheet, hopefully you will find the information you need:
https://cdn.sparkfun.com/datasheets/Sensors/ForceFlex/hx711_english.pdf

as far as i'm aware, there's no "slave address" in hx711, not in the i2c sense anyway, if that's what you're asking.

YudiNz, 2021-04-27:
> i don't think i understand your question. however, here's a link to the datasheet, hopefully you will find the information you need: [...]

I was asking in the i2c sense way due to it being describes are working over i2c. So is it just basically always reading the clock and data lines to see if they single data is ready to be transmitted?

Thanks for your reply :D

bogde, 2021-04-27:
yes, the chip is not using any standard protocols as far as i'm aware, so no i2c, it's basically a custom protocol, as described in the datasheet. so yes, you just have to use the clock and data lines to get readings out of the chip.

YudiNz, 2021-04-27:
> yes, the chip is not using any standard protocols as far as i'm aware, so no i2c, it's basically a custom protocol, as described in the datasheet. so yes, you just have to use the clock and data lines to get readings out of the chip.

Thanks so much for clearing this up for me. Makes it easier for me to get it working in LabVIEW myRio

closed 2021-04-27 by YudiNz