bogde/HX711 #207: Add support for Nordic NRF52840 microcontroller

ElecSmurf, 2021-07-23:
Dear,

the module does not seem to work on arduino nano 33 ble (sense) which is build on the Nordic NRF52840 microcontroller.
Is it possible to add support for this microcontroller?

Regards

amotl, 2021-07-24:
Hi there,

thanks for writing in. Can you elaborate a bit more what exactly doesn't work? Does your code compile or show any errors on it? Do you observe any failures or misbehaviour at runtime?

With kind regards,
Andreas.

ElecSmurf, 2021-07-25:
Hi Andreas,

Thanks for the swift response!
The code compiles without errors, however, the following simple sketch:

#include <HX711.h>
#define calibration_factor -7050.0
#define DOUT 2
#define CLK 3
#define DEBUG
HX711 scale;
void setup() {
Serial.begin(9600);
Serial.println("Started");
scale.begin(DOUT, CLK);
scale.set_scale(calibration_factor);
scale.tare();
}
void loop() {
Serial.println(-2.1 * scale.get_units(), 1); //scale.get_units() returns a float
}

Thsi sketch hangs on the scale.tare(); line

As I don't see any support in the code for the nano33ble (nordicnrf52) platform
https://docs.platformio.org/en/latest/boards/nordicnrf52/nano33ble.html
I forced the FAST_CPU path, but this dit not resolve the issue.
The code hangs on 'wait_ready', replacing with wait_ready_retry(10) it no longer hangs, but does not provide valid value's.

On Arduino uno the same setup works perfectly, but I would like to have it working on the nano 33 ble so I can make use of the bleutooth feature.

Regards,

Kris

amotl, 2021-07-25:
Hi Kris,

thanks for reporting more details about your observations. Apparently, support for the nRF52840 has been added by @bogde through 6f8705f9e1f4 the other day. Maybe he can add something valuable to this discussion?

But you are right, I also don't see this MCU being explicitly defined as `FAST_CPU`, like the others in this list:
https://github.com/bogde/HX711/blob/8ca6bb665959832f2f44764840993fc980d84ea6/src/HX711.cpp#L24-L29

However, isn't this chip of a SAM or SAMD flavor?

> I forced the `FAST_CPU` path, but this did not resolve the issue.

Thanks already for trying this. It looks like you already took a reasonable dive into the code base, I appreciate that.

Did you notice that the signature of the `tare` function is actually `void HX711::tare(byte times)`? According to your code, you are invoking the function without any parameter, which means it will default to "10" actual readings. Maybe it will work if you reduce that to, e.g., "1"?

Maybe you can also try to reduce `board_build.f_cpu = 64000000L` in order to check if the MCU speed would be still too fast?

That's for now the things which immediately came to my mind in order to approach the problem. I dearly hope that a solution can be found in order to add the "Arduino Nano 33 BLE" to the official list of verified boards. Please let us know about any progress you might be able to make.

With kind regards,
Andreas.

bogde, 2021-07-26:
hello, i only tested the code on the adafruit feather and it worked. also, no fancy stuff, just the basic example.

@ElecSmurf is this the board you're trying to use: 
https://www.tme.eu/de/en/details/abx00030/arduino-solutions/arduino/arduino-nano-33-ble/

thanks!

ElecSmurf, 2021-07-26:
@bodge that's correct

https://www.tme.eu/de/en/details/abx00030/arduino-solutions/arduino/arduino-nano-33-ble/

bogde, 2021-07-28:
are you sure your wiring is correct? i bought an arduino nano 33 ble and your sketch works out of the box, see my images.

[image]

[image]

[image]

i added a delay after serial.begin but it works without that too.

bogde, 2021-07-28:
and yes, the sketch does hang on the scale.tare() if the hx711 module is not connected to pins D2 and D3.

ElecSmurf, 2021-07-29:
@bogde a big thanks for the support and effort!!! My appoligies, it was indeed a faulty connectivity which I checked over and over but must have been tunnel vision from my side. All is working correctly now!

bogde, 2021-07-29:
glad it worked! im closing this now, feel free to open it again or create a new ticket if you have any other questions. cheers!

closed 2021-07-29 by bogde