bogde/HX711 #208: Need The Raw Data To Be A Voltage

ariqhwr, 2021-07-29:
Hi i am currently working on a project using a load cell and the HX711 as the amplifier. And i found out about this library and the example code from it (attached below). And i currently face some problems now.

1. The raw data that i got (from scale.read_average(20)) was in the range between -10000 to about 20,000,000 (use 3 load cells and the result more or less the same). Isn't supposed to be between -8388608 to 8388608? Since the output of HX711 is 2^24 bit? Or are there something that i missed?
2. Long story short, i need to convert the data that i got to become a voltage, but i really don't know how to convert this data to be a voltage that i needed. I really need some help with this, are there any equation, example, and/or literature that i can read? Thank you in advance!

Code:
#include "HX711.h"

// HX711 circuit wiring
const int LOADCELL_DOUT_PIN = 4;
const int LOADCELL_SCK_PIN = 5;

HX711 scale;

void setup() {
  Serial.begin(115200);
  Serial.println("HX711 Demo");

  Serial.println("Initializing the scale");

  // Initialize library with data output pin, clock input pin and gain factor.
  // Channel selection is made by passing the appropriate gain:
  // - With a gain factor of 64 or 128, channel A is selected
  // - With a gain factor of 32, channel B is selected
  // By omitting the gain factor parameter, the library
  // default "128" (Channel A) is used here.
  scale.begin(LOADCELL_DOUT_PIN, LOADCELL_SCK_PIN);

  Serial.println("Before setting up the scale:");
  Serial.print("read: \t\t");
  Serial.println(scale.read());			// print a raw reading from the ADC1

  Serial.print("read average: \t\t");
  Serial.println(scale.read_average(20));  	// print the average of 20 readings from the ADC

  Serial.print("get value: \t\t");
  Serial.println(scale.get_value(5));		// print the average of 5 readings from the ADC minus the tare weight (not set yet)

  Serial.print("get units: \t\t");
  Serial.println(scale.get_units(5), 1);	// print the average of 5 readings from the ADC minus tare weight (not set) divided
						// by the SCALE parameter (not set yet)

  scale.set_scale(414.f);                      // this value is obtained by calibrating the scale with known weights; see the README for details
  scale.tare();				        // reset the scale to 0

  Serial.println("After setting up the scale:");

  Serial.print("read: \t\t");
  Serial.println(scale.read());                 // print a raw reading from the ADC

  Serial.print("read average: \t\t");
  Serial.println(scale.read_average(20));       // print the average of 20 readings from the ADC

  Serial.print("get value: \t\t");
  Serial.println(scale.get_value(5));		// print the average of 5 readings from the ADC minus the tare weight, set with tare()

  Serial.print("get units: \t\t");
  Serial.println(scale.get_units(5), 1);        // print the average of 5 readings from the ADC minus tare weight, divided
						// by the SCALE parameter set with set_scale

  Serial.println("Readings:");
}

void loop() {
  Serial.print("one reading:\t");
  Serial.print(scale.get_units(), 1);
  Serial.print("\t| average:\t");
  Serial.println(scale.get_units(10), 1);

  scale.power_down();			        // put the ADC in sleep mode
  delay(5000);
  scale.power_up();
}

ba05, 2021-07-31:
Read the raw LSBs returned, then:
`mV/V = rawLSBs / maxLSBs * ADC_diff_ratio/ ADCgain * 1000`
where:
```
maxLSBs = 8388608
ADC_diff_ratio = 0.5
ADCgain = 128
```

ba05, 2021-07-31:
I think you can get mV/V like this:
`mV/V = rawLSB / maxLSB * ADC_diff_ratio / ADCgain * 1000`
where:
`rawLSB` is the count value returned by the HX711
`maxLSB = 8388608`
`ADC_diff_ratio  = 0.5`
`ADCgain = 0.5`

ariqhwr, 2021-08-01:
Thank you for the answer! Are there any particular literature/website/book that i could read as my source to learn about this?

Also, what does the ADCgain means? Isn't it suppose to be 128/64/32 depends on what i used?

Thank you

ariqhwr, 2021-08-01:
And also, what is ADC diff ratio? I try to search it anywhere but i can't find it

ba05, 2021-08-01:
> Thank you for the answer! Are there any particular literature/website/book that i could read as my source to learn about this?
> 
> Also, what does the ADCgain means? Isn't it suppose to be 128/64/32 depends on what i used?
> 
> Thank you

Yes gain is 128 (I had a typo, fixed). ADC diff ratio accounts for the +/- of the voltage (both sides of zero). 

Also note it looks like the actual mV/V value is going to have an offset of a percent or two due to gain error offset and other factors. Still best to perform a calibration. I think the offset should be fairly constant.

See the TI ADS1231 data sheet. [TI ADS1231](https://www.ti.com/lit/ds/symlink/ads1231.pdf). Different device yes, but it explains some common problems things better than the HX711 data sheet.

ariqhwr, 2021-08-01:
Ahh i see. Thank you for the answer.

And yes i also realize the existance of the offset after the calculation and everything, definitely do a calibration after this.

After i read the data sheet i couldn't find the equation that you mentioned before, or did i missed it?

ba05, 2021-08-04:
I derived the equation using a table in the TI data sheet.

You probably want to keep everything as mV/V not voltage (V). If you have to convert to voltage, multiply the mV/V by your excursion voltage.

ariqhwr, 2021-08-09:
Im just curious, are there any correlations between reading the HX711 (which is an 24 bit ADC) using Arduino Uno (10-bit ADC) and the data that i got? Which isn't near the specification in the datasheet of HX711 (mine is about 0-2000000, whereas the datasheet said it should hover around -8388608 to 8388608)

Or is there any other reasons why? I am new at this

Thanks for the help!