bogde/HX711 #209: Using INB to measure excitation voltage

mcmahad, 2021-07-30:
HX711 comes with two analog channels.  As best I can tell, INB is always ignored in applications.

It would seem like this is the perfect way to monitor the excitation voltage.   Load cells return values in mV/Volt, but I think most people skip this relationship.  It means that if your excitation voltage drifts over time (regulators aren't perfect), then your scaling will be off by some amount.  Wouldn't it be better to measure excitation voltage at the time INA is measured so the ratio can be calculated before scaling and offset is applied?

Of course, excitation voltage is much too large to measure directly with INB.  It would need to be scaled down by about 10x or more to be range of the INB channel.  That can be done with two resistors.  You don't even need really accurate resistors, since we don't need to work in ideal volts.   Tare and scale calibration of the final circuit will just contain a small correction for the resistor values baked into scaling constants.

By doing this, I would hope that calibration would remain accurate for longer periods and over wider temperature ranges.   The B channel is typically unused anyway, so might as well do something with it. 

In the ideal world, you would do this with a six-wire load cell and use the extra two wires to make a Kelvin connection to the load cell.    This removes error due to the voltage drop across excitation lines that carry current to the load cell.

 Any thoughts?

ba05, 2021-07-31:
See my recent reply here [mV/V output](https://github.com/bogde/HX711/issues/208#issuecomment-890372140)

ba05, 2021-07-31:
See my other comment here [mV/V output on HX711](https://github.com/bogde/HX711/issues/208#issuecomment-890373362)