bogde/HX711 #210: 0.7.4 seems a bit broken

lalitkumarj, 2021-08-14:
Hi All

Using 0.7.4 in Arduino IDE I really struggled to calibrate. In fact, when I put two weights on, A<B, the resulting outputs satisfied read(A) > read(B)!

I just installed 0.7.2 and things seem to be working better. Just wanted to call this to attention.

lalitkumarj, 2021-08-14:
Ok, looks like this line changed: 
https://github.com/bogde/HX711/blob/bc5a7e5a0d518a0f384bd30105cdce0773e20028/src/HX711.cpp#L76

pinMode(DOUT, INPUT) ----> pinMode(DOUT, INPUT_PULLUP);

I am on an ESP8266. Maybe this is not compatible?

amotl, 2021-09-14:
Hi there,

thank you for reporting this. It might also be related to #193.

With kind regards,
Andreas.