bogde/HX711 #216: This issue still persists. I think that's a fault in the chip design itself.

jagannathmurali, 2021-11-19:
This issue still persists. I think that's a fault in the chip design itself.

```
170.75
170.83
171.22
170.80
170.95
171.04
-143.37
171.36
170.72
170.86
171.10
170.81
170.90
171.05
170.81
```

EDIT: What I've discovered, however, is that when a spike arises, the **next** ADC value needs a longer time than usual. This could be used to filter out _most_ of these spikes.

```
0.03 - 4ms
0.05 - 3ms
0.02 - 3ms
1302.99 - 4ms
0.06 - 10ms
0.05 - 3ms
0.01 - 4ms
...

```

_Originally posted by @Fusseldieb in https://github.com/bogde/HX711/issues/83#issuecomment-844327053_

jagannathmurali, 2021-11-19:
I also get some spike

Fusseldieb, 2021-11-19:
I do think it's an issue in how Write/Read cycles are handled by the Arduino/ESP. It chokes on itself from time to time...

X-Ryl669, 2021-11-24:
You can median filter those easily:
```
   static inline void swap(float &a, float &b) { float t = a; a = b; b = t; }
   static inline float medianFilter(float value) {
      static float acc[3] = {0, 0, 0};
      static int pos = 0;
      acc[pos % 3] = value;
      if (pos == 0) return value;
      pos++;
      if (acc[0] > acc[1]) swap(acc[0], acc[1]);
      if (acc[1] > acc[2]) swap(acc[1], acc[2]);
      return acc[1];
   }
```

CossartSim, 2024-01-19:
I found a workaround for this issue. You can simply power down the chip between each reading. It removes all the spikes for me but significantly reduces the acquisition frequency.