bogde/HX711 #217: Load cell value consistently rising and falling despite no load being applied.

diiarmuid, 2021-11-24:
Hi, I've been having this persistent issue when using the HX711 amplifier with the 50kg load cell. Basically on calibration or testing of load cell, the value I get back is consistently rising and falling even though there is no weight being applied. When I apply weight the value changes to match the applied force but the consistent rising and falling of the value still happens when the load is both applied and not applied.

The screenshot bellow highlights the issue.

I've tried using multiple amplifiers and multiple different libraries but the issue persists, as well as changing the load cells but again the issue remains.

I'm currently using an Arduino Uno if that's relevant in any way.

Would really appreciate any support as I'm at wits end with this atm and cant seem to find any solution online for this issue.

Thank you.

[image]

wranglatang, 2022-07-04:
Hello @diiarmuid, i have the exact same issue at the moment, Using ESPHome for my load cells and i see this rising and falling despite no load being applied. Did you ever find out how to solve this?

Mausbiber, 2022-09-11:
same problem here

JimBO-3D, 2023-08-28:
I have the same problem. Did you find solution?

Sjeggefjes, 2023-08-29:
Hi, I did not have this problem and this leads me to think there might be a poor connection between the loadcell and the amplifier or electrical noise issues. Try to resolder the connections or change the wires between the amplifier and the loadcells to see if this is the case.