bogde/HX711 #222: is_ready function broken for espressif microcontrollers after #193

ayushsharma82, 2021-12-16:
Hi @bogde & @Simonas0 ,

The ‘is_ready’ function of library is broken for Espressif chips (ESP32 & ESP8266) after merger of #193 pull request. Is there any other way by which we can detect if HX711 is present? 

Thanks,
Ayush

bogde, 2021-12-21:
so you mean setting DOUT to INPUT breaks the `is_ready` function?
did you test by editing src/hx711.cpp and changing 
`#define DOUT_MODE INPUT` 
under `#ifdef ARCH_ESPRESSIF`
to 
`#define DOUT_MODE INPUT_PULLUP`
and it worked?

ayushsharma82, 2021-12-21:
I did some more research, the function works fine the HX711 is connected with MCU.

In my application, the HX711 is a pluggable external module, so I'm particularly using the `is_ready` function to detect if HX711 is connected to MCU or not. The problem is due to the `is_ready` function checking for DOUT line to be LOW. With the recent merge of #193, the DOUT line remains floating when **no** HX711 board is connected hence the false value of `true` is returned.

**Possible Solution:**
We can have an `is_connected` function that checks the DOUT line for a short period with `INPUT_PULLUP`?

ayushsharma82, 2021-12-21:
> so you mean setting DOUT to INPUT breaks the `is_ready` function? did you test by editing src/hx711.cpp and changing `#define DOUT_MODE INPUT` under `#ifdef ARCH_ESPRESSIF` to `#define DOUT_MODE INPUT_PULLUP` and it worked?

Tested with v0.7.4, that was before #193, so technically yes. The `is_ready` function works fine when used with INPUT_PULLUP to detect if the module is present or not.

Ben17028, 2022-03-21:
is_ready doesn´t work correctliy in 0.7.5 @ andurino UNO. Software pullup the DOUT pin does not fix it. 
In 0.7.4 it works fine.

andreash-esp, 2022-08-09:
The wait_ready_timeout() function is broken as well in 0.7.5 for ESP32.

This code triggers "HX711 detected" even when no HX711 is attached:
`
  scale.begin(hx711_dt_pin, hx711_sck_pin);
  if (scale.wait_ready_timeout(1000)) {               // check for HX711
    scale.power_up();
    Serial.println("HX711 detected");
  }
`
The code is working fine in 0.7.4. Build environment is arduino-esp32 1.0.6