bogde/HX711 #229: using 4 load cells

Kowng, 2022-02-17:
Hi bodge i was wondering how if there is a code that takes the average of 4 load cells and gives me a single figure. Thanks.

tvdb2003, 2024-12-23:
I think this will work if you make multiple instance's of hx711 scale for example something like this:
(this is only for 2 loadcells so you should extend it if you would like more)

`/**
 *
 * HX711 library for Arduino - example file
 * https://github.com/bogde/HX711
 *
 * MIT License
 * (c) 2018 Bogdan Necula
 *
**/
#include "HX711.h"

// HX711 circuit wiring
const int LOADCELL_DOUT_PIN0 = 2;
const int LOADCELL_SCK_PIN0 = 3;

const int LOADCELL_DOUT_PIN1 = 4 ;// or whatever pin you want to use
const int LOADCELL_SCK_PIN1 = 5; // or whatever pin you want to use
long average_weight = 0;
HX711 scale;
HX711 scale1; // do this as well for load cell 3 and 4

void setup() {
  Serial.begin(38400);
  Serial.println("HX711 Demo");

  Serial.println("Initializing the scale");

  // Initialize library with data output pin, clock input pin and gain factor.
  // Channel selection is made by passing the appropriate gain:
  // - With a gain factor of 64 or 128, channel A is selected
  // - With a gain factor of 32, channel B is selected
  // By omitting the gain factor parameter, the library
  // default "128" (Channel A) is used here.
  scale.begin(LOADCELL_DOUT_PIN0, LOADCELL_SCK_PIN0);
  scale1.begin(LOADCELL_DOUT_PIN1, LOADCELL_SCK_PIN1);

  Serial.println("Before setting up the scale:");
  Serial.print("read: \t\t");
  Serial.println(scale.read());			// print a raw reading from the ADC

  Serial.print("read average: \t\t");
  Serial.println(scale.read_average(20));  	// print the average of 20 readings from the ADC

  Serial.print("get value: \t\t");
  Serial.println(scale.get_value(5));		// print the average of 5 readings from the ADC minus the tare weight (not set yet)

  Serial.print("get units: \t\t");
  Serial.println(scale.get_units(5), 1);	// print the average of 5 readings from the ADC minus tare weight (not set) divided
						// by the SCALE parameter (not set yet)

  scale.set_scale(2280.f);                      // this value is obtained by calibrating the scale with known weights; see the README for details
  scale.tare();				        // reset the scale to 0

  scale1.set_scale(2280.f);                      // this value is obtained by calibrating the scale with known weights; see the README for details
  scale1.tare();				        // reset the scale to 0

  Serial.println("After setting up the scale:");

  Serial.print("read: \t\t");
  Serial.println(scale.read());                 // print a raw reading from the ADC

  Serial.print("read average: \t\t");
  Serial.println(scale.read_average(20));       // print the average of 20 readings from the ADC

  Serial.print("get value: \t\t");
  Serial.println(scale.get_value(5));		// print the average of 5 readings from the ADC minus the tare weight, set with tare()

  Serial.print("get units: \t\t");
  Serial.println(scale.get_units(5), 1);        // print the average of 5 readings from the ADC minus tare weight, divided
						// by the SCALE parameter set with set_scale

  Serial.println("Readings:");
}

void loop() {
average_weight = (scale.get_units(),1 + scale1.get_units(),1)/2;
Serial.print("average weight:" );
Serial.println(average_weight);
  scale.power_down();			        // put the ADC in sleep mode
  delay(5000);
  scale.power_up();
}
`

but i only have one loadcell so you should try it yourself to see if it works.

tvdb2003, 2024-12-23:
``/**
 *
 * HX711 library for Arduino - example file
 * https://github.com/bogde/HX711
 *
 * MIT License
 * (c) 2018 Bogdan Necula
 *
**/
#include "HX711.h"

// HX711 circuit wiring
const int LOADCELL_DOUT_PIN0 = 2;
const int LOADCELL_SCK_PIN0 = 3;

const int LOADCELL_DOUT_PIN1 = 4 ;// or whatever pin you want to use
const int LOADCELL_SCK_PIN1 = 5; // or whatever pin you want to use
long average_weight = 0;
HX711 scale;
HX711 scale1; // do this as well for load cell 3 and 4

void setup() {
  Serial.begin(38400);
  Serial.println("HX711 Demo");

  Serial.println("Initializing the scale");

  // Initialize library with data output pin, clock input pin and gain factor.
  // Channel selection is made by passing the appropriate gain:
  // - With a gain factor of 64 or 128, channel A is selected
  // - With a gain factor of 32, channel B is selected
  // By omitting the gain factor parameter, the library
  // default "128" (Channel A) is used here.
  scale.begin(LOADCELL_DOUT_PIN0, LOADCELL_SCK_PIN0);
  scale1.begin(LOADCELL_DOUT_PIN1, LOADCELL_SCK_PIN1);

  Serial.println("Before setting up the scale:");
  Serial.print("read: \t\t");
  Serial.println(scale.read());			// print a raw reading from the ADC

  Serial.print("read average: \t\t");
  Serial.println(scale.read_average(20));  	// print the average of 20 readings from the ADC

  Serial.print("get value: \t\t");
  Serial.println(scale.get_value(5));		// print the average of 5 readings from the ADC minus the tare weight (not set yet)

  Serial.print("get units: \t\t");
  Serial.println(scale.get_units(5), 1);	// print the average of 5 readings from the ADC minus tare weight (not set) divided
						// by the SCALE parameter (not set yet)

  scale.set_scale(2280.f);                      // this value is obtained by calibrating the scale with known weights; see the README for details
  scale.tare();				        // reset the scale to 0

  scale1.set_scale(2280.f);                      // this value is obtained by calibrating the scale with known weights; see the README for details
  scale1.tare();				        // reset the scale to 0

  Serial.println("After setting up the scale:");

  Serial.print("read: \t\t");
  Serial.println(scale.read());                 // print a raw reading from the ADC

  Serial.print("read average: \t\t");
  Serial.println(scale.read_average(20));       // print the average of 20 readings from the ADC

  Serial.print("get value: \t\t");
  Serial.println(scale.get_value(5));		// print the average of 5 readings from the ADC minus the tare weight, set with tare()

  Serial.print("get units: \t\t");
  Serial.println(scale.get_units(5), 1);        // print the average of 5 readings from the ADC minus tare weight, divided
						// by the SCALE parameter set with set_scale

  Serial.println("Readings:");
}

void loop() {
average_weight = (scale.get_units(),1 + scale1.get_units(),1)/2;
Serial.print("average weight:" );
Serial.println(average_weight);
  scale.power_down();			        // put the ADC in sleep mode
  delay(5000);
  scale.power_up();
}