bogde/HX711 #244: Should function "scale.tare();" need a parameter?

mianqi2016, 2022-09-03:
I tested the example "HX711_full_example", in line 49, it was:
`scale.tare();				        // reset the scale to 0`
I found it was same in example on github, my question are:
1. due to the function's prototype in header file, there should be a parameter in the calling, even it was "0", and not be null? 
`void HX711::tare(byte times) {
	double sum = read_average(times);
	set_offset(sum);
}
2. since the function's prototype has "set_offset(sum);" inside, so the comment should be "reset the offset to 0"?

ThierryChatGarou, 2022-10-12:
For 1: A default arguments was defined as void tare(byte times = 10); in the header file. So even if you call scale.tare(); with no argument still valid.
For 2: I believe the current comment is correct. Calling Tare sets the scale to Zero, but doesn't mean the OFFSET is set to Zero.
The subtraction of the read value minus the OFFSET is what makes your get_value() or get_units() function give zero.

banoz, 2023-05-14:
Currently there's an issue with the tare method described [here](https://github.com/bogde/HX711/issues/228#issuecomment-1079080810).