bogde/HX711 #246: Verify my wiring (HX711 with four strain gauges)

JustApu, 2022-12-23:
[image]
[image]

I'm trying to verify all my wiring is correct. Disconnected from the HX711, I'm getting

E- to A+ is 2kΩ
E- to A- is 4kΩ
E- to E+ is 6kΩ

Does that sound right?

Connected to the HX711, I'm getting a constant 8388607 regardless of weight applied to the platform.

_(Side note: The "F" & "M" designations on the schematic are female & male Dupont connectors which join the various strain gauges together. My way of making it easier to have multiple test points and re-wire as necessary though, obviously, also potential failure points as well.)_

TheGermanUSA, 2023-01-05:
Yes, that is exactly how I have it set up and it works fine.

JustApu, 2023-01-05:
Thanks, @TheGermanUSA - wondering what is wrong with my setup then.