bogde/HX711 #251: HX711 only reads once on startup

Xarbenence, 2023-04-29:
Hello,

I wanted to post a question regarding a project I am working on. I am working with 4. I have an HX711 interfacing with 4 half bridge strain gauges: [found here](https://www.amazon.com/Weighing-Resistance-Half-Bridge-Amplifier-Arduino/dp/B097T3SX6W/ref=sr_1_6?crid=15P6KG8R7OUKZ&keywords=strain+gauge+load+cell&qid=1682793072&sprefix=strain+gauge+load+cell%2Caps%2C100&sr=8-6). When I use the library and initialize the board, I have a loop in my setup() that just checks to see if the board is ready to be read from. I never get stuck in that loop, and I can successfully read from the board immediately after doing scale.begin(DOUT,SCL); However, when I enter my actual loop(), I scale.is_ready() always returns false. I even tested the first reading I get with different weights, and the value changes, telling me that the load cells/HX711 are working, but I'm at a loss for why I need can only read once. In my loop(), if I re-call scale.begin(), I can get more reads, but I don't want to have to reset the HX711 every time I want to read from it. Has anyone else experienced this issue? Thanks.

banoz, 2023-05-14:
Did you try to call `scale.power_up()` after `scale.begin()`?

fenixil, 2024-02-23:
@Xarbenence 
I hit the same issue today, so let me explain what's likely causing yours.
First, here is the link to the datasheet - https://www.digikey.com/htmldatasheets/production/1836471/0/0/1/hx711.html
As you can see, module can provide 10 or 80 samples per second. It needs some time to provide the value after it's started, that's why your first loop always succeeds as you wait for 1 second between iterations.
Your second loop does not have any delays, so it will exit almost immediately. You must add a delay there too.

```
weight = scale.get_units(); 
// ^^^^ you read the data here, so is_ready() becomes false
...
while(i<9){
if(scale.is_ready()){ // this is_ready() always fails
  long reading = scale.read();
  Serial.print("HX711 reading: ");
  Serial.println(reading);
  break;
} else {
  delay(100);
  // ^^^^ here
}
```

As I can see from the sources, `begin()` does not perform communication with the module, so it's not something that helps you here, but rather the delay at the end of the script, or something else.

I had and identical issue cause by `power_up` and `power_down` - I called power_up and expected data was available immediately. That was obviously wrong, so after I added some delay (just hardcoded 500ms to comply with Output settling time) everything got back to normal.

PS. I'd greatly appreciate if you edit your post, use github's markdown to make it look like a code and clean it up from commented code.