bogde/HX711 #254: What are possible values for  scale.set_scale(2280.f);

Peter-Stuhr, 2023-06-06:
I have not found any information about the meaning of "2280.f"
Can anybody explain to me, what happens, if I modify this number? What are the alternatives?
I need kg for my application.

Thanks.

amotl, 2023-06-06:
Dear Peter,

in order to learn about the specific value of your load cell, you will need to put a known weight on it, and run through the procedure outlined in the README at [^1].

If you like to explore an interactive variant of that load cell adjustment procedure, on behalf of a ready-made program/sketch, you are welcome to have a look at [^2][^3] from the Hiveeyes project.

With kind regards,
Andreas.

/cc @ClemensGruber

[^1]: https://github.com/bogde/HX711/tree/master#how-to-calibrate-your-load-cell
[^2]: https://hiveeyes.org/docs/arduino/firmware/openhive/scale-adjust/README.html
[^3]: https://github.com/hiveeyes/arduino/blob/main/openhive/scale-adjust/HX711/scale-adjust-hx711.ino

Peter-Stuhr, 2023-06-07:
Hallo Andreas,
danke für die schnelle Info. Ich habe den Wert 2280.f gelassen und die Waage hat exakt mein Gewicht angezeigt. Ich weiß jetzt zwar immer noch nicht, wie genau dieser Wert zustande gekommen ist, aber er stimmt wohl.

Gruß Peter