bogde/HX711 #258: HX711 high current consumption

nobodyguy, 2024-01-05:
Hello,
have anyone experienced high current draw when the HX711 is powered down and the analog supply voltage is below 5V?
More details are here - https://electronics.stackexchange.com/questions/696330/hx711-analog-supply-regulator-current-draw-in-sleep-mode

pomplesiegel, 2025-01-23:
Yes! We have discovered that using this library we sometimes have significantly higher current consumption during power down. From testing we found that with zero code change but a swap from the newer HX711 part to an older one the problem goes away. 

I'm trying to see if there are any workarounds other that sourcing the older HX711 chip. Perhaps a different library? 

AVIA says the chips are identical, but clearly they are not. FYI even when power_down has been engaged properly on the new HX711 part, it still draws an additional 5-8uA. Something is wrong here, as that's not consistent with the datasheet. 

Any ideas??

Old part (says only HX711) - 10uA total system sleep current from 3V supply - PASS

[image]

New part (has "AVIA" marking) - 60+uA total system sleep current from 3V supply - FAIL

[image]

nobodyguy, 2025-01-23:
On this issue I've learned to never ever trust any cheap part :)
So I did a new HW revision with a mosfet power switch and I power off the whole chip when the main MCU is sleeping (and I disconnect all the related GPIOs).

pomplesiegel, 2025-01-23:
Nice solution! Anyone else experiencing this high current during issue? I've seen it on 100% of the "new parts" (above) I've tried. 

I'm going to try the [adafruit HX711 driver ](https://github.com/adafruit/Adafruit_HX711/tree/main) next to see if that makes any difference.

nobodyguy, 2025-01-23:
Depending on the implementation, it may also be worth checking a logic analyzer to verify if the SCK pulses are correct, ensuring that interrupt locking works as expected.
I encountered this issue https://devzone.nordicsemi.com/f/nordic-q-a/113092/gpio-bit-banging-and-ble-radio-activity, which caused power down/up events and increased current consumption.

pomplesiegel, 2025-01-29:
Thank you @nobodyguy! Update from my side: With particular IC samples I'm still intermittently seeing much higher sleep / power_down currents, even when I add delayMicroseconds() or disable interrupts to ensure the SCK pulses are being handled correctly. 

About half of the time the sleep current is as described - <1uA, but the other half it's more than 10uA, floating around. Anyone else seeing this? 

I've tried many variations around timing for the power_down() method and it has not made a difference in these chips' behavior. 

BTW, now i'm seeing it on some chips with the standard HX711 silk above (old part).

pomplesiegel, 2025-01-29:
I'm going to try the [adafruit HX711 driver ](https://github.com/adafruit/Adafruit_HX711/tree/main) and it showed the exact same intermittent high current behavior, so I don't think the root cause is the driver. However, perhaps it could be fixed by a workaround in the driver. 
Anyone else experiencing something like this? Based on my sample it must be fairly widespread